LibreCAD/LibreCAD #575: LibreCAD 2.0.8: section [RecentFiles] is not written to LibreCAD.conf

jborme, 2015-09-22:
Since I upgraded to libreCAD 2.0.8, the recent files are not updated in the configuration file. The section [RecentFiles] is read when present in the configuration file and the File menu in the GUI is populated with the entries. The File menu in the GUI is also updated as long as LibreCAD is running. However, when exiting the software, the [RecentFiles] section is not written back to the configuration file (if existing), and the section is not created if the configuration file is created anew.

I am using LibreCAD 2.0.8 on gentoo linux.

Version : master
Revision : 2.0.8
Compiler: GNU GCC 4.8.5
Qt Version: 5.4.2
Compiled on: Sep 8 2015

dxli, 2015-09-23:
Can not reproduce this bug.

jborme, 2015-09-23:
I compiled again libreCAD 2.0.7 and 2.0.8. I specifically get the bug in 2.0.8 and not in 2.0.7. The behaviour is the same when linking to qt 4.8.6 or qt 5.4.2.

dxli, 2015-09-23:
please try to remove the old conf file and try again

jborme, 2015-09-23:
rm ~/.config/LibreCAD -rf
I then launched LibreCAD. In the software, I saved a file. It appears in the menu (attached image).
After exiting libreCAD, here are the complete contents of the file 

~/.config/LibreCAD/libreCAD.conf

[Appearance]
Language=en
LanguageCmd=en

[Defaults]
Unit=Millimeter

[Print]
PaperSizeX=210
PaperSizeY=297

[Paths]
Save=/home/jerome/init

[Geometry]
WindowWidth=950
WindowHeight=700
WindowX=889
WindowY=212
DockWindows=@ByteArray(\0\0\0\xff\0\0\0\0\xfd\0\0\0\x3\0\0\0\0\0\0\0\0\0\0\0\0\xfc\x2\0\0\0\x1\xfb\0\0\0\x12\0L\0i\0\x62\0r\0\x61\0r\0y\0\x44\0W\0\0\0\0\0\xff\xff\xff\xff\0\0\0\xd9\x1\0\0\x5\0\0\0\x1\0\0\x1\n\0\0\x1\xc8\xfc\x2\0\0\0\x2\xfb\0\0\0\xe\0L\0\x61\0y\0\x65\0r\0\x44\0W\x1\0\0\0Q\0\0\0\xe3\0\0\0\xde\x1\0\0\x5\xfb\0\0\0\xe\0\x42\0l\0o\0\x63\0k\0\x44\0W\x1\0\0\x1\x37\0\0\0\xe2\0\0\0\x9a\x1\0\0\x5\0\0\0\x3\0\0\x3m\0\0\0\x80\xfc\x1\0\0\0\x1\xfb\0\0\0\x12\0\x43\0o\0m\0m\0\x61\0n\0\x64\0\x44\0W\x1\0\0\0I\0\0\x3m\0\0\x1\v\x1\0\0\x5\0\0\x2`\0\0\x1\xc8\0\0\0\x4\0\0\0\x4\0\0\0\b\0\0\0\b\xfc\0\0\0\x3\0\0\0\0\0\0\0\x1\0\0\0\n\0\x43\0\x41\0\x44\0T\0\x42\x3\0\0\0\0\xff\xff\xff\xff\0\0\0\0\0\0\0\0\0\0\0\x2\0\0\0\x4\0\0\0\f\0S\0n\0\x61\0p\0T\0\x42\x1\0\0\0\0\xff\xff\xff\xff\0\0\0\0\0\0\0\0\0\0\0\f\0\x46\0i\0l\0\x65\0T\0\x42\x1\0\0\x1\xa6\xff\xff\xff\xff\0\0\0\0\0\0\0\0\0\0\0\f\0\x45\0\x64\0i\0t\0T\0\x42\x1\0\0\x2\x84\xff\xff\xff\xff\0\0\0\0\0\0\0\0\0\0\0\f\0Z\0o\0o\0m\0T\0\x42\x1\0\0\x3Q\xff\xff\xff\xff\0\0\0\0\0\0\0\0\0\0\0\x2\0\0\0\x2\0\0\0\n\0P\0\x65\0n\0T\0\x42\x1\0\0\0\0\xff\xff\xff\xff\0\0\0\0\0\0\0\0\0\0\0\f\0T\0o\0o\0l\0T\0\x42\x1\0\0\x1\xa4\xff\xff\xff\xff\0\0\0\0\0\0\0\0)

[image]

lordofbikes, 2015-09-23:
I've checked both versions on Ubuntu too, without any issue. A new file is added to the Recent Files with both latest versions, master (2.0.7) and 2.0 (2.0.8) branch.

r-a-v-a-s, 2015-10-13:
It works on Windows 7.

I'm wondering if it is because we use a `/` at the beginning of section names.

`RS_SETTINGS->beginGroup("/RecentFiles");`

http://doc.qt.io/qt-5/qsettings.html#section-and-key-syntax

> Do not use slashes ('/' and '\') in section or key names; the backslash character is used to separate sub keys (see below). On windows '\' are converted by QSettings to '/', which makes them identical.

Their example:
`settings.beginGroup("mainwindow");`

r-a-v-a-s, 2016-09-07:
2.1.1 is available.
https://packages.gentoo.org/packages/media-gfx/librecad

Let us know if the problem still exists after upgrading.

jborme, 2016-09-08:
The problem is solved in version 2.1.1.

closed 2016-09-08 by jborme