LibreCAD/LibreCAD #576: can't see preview with certain combinations of background color and pen color

r-a-v-a-s, 2015-09-23:
If I set the background to white and I have a black pen, then I can not see the preview when drawing.

dxli, 2015-09-23:
should be an easy one.

I still don't quite understand why CAD likes a smart "black/white" color. If we set it to two black or white options.
User can ask for white background and white pen, and a blank page won't be considered as a bug

r-a-v-a-s, 2015-10-03:
This is why we can't see the preview:
https://github.com/LibreCAD/LibreCAD/commit/fba9948b6fe21f33ef5e98ca8c7136adab54aeca

r-a-v-a-s, 2015-10-08:
The composition mode breaks the principle of least astonishment.
People will keep posting about this.
I don't think being able to see orthogonal lines under the crosshair is worth it.
I suggest removing this in favor of an additional crosshair as I have suggested.

r-a-v-a-s, 2015-10-10:
I'll work on this.

closed 2015-10-11 by r-a-v-a-s