LibreCAD/LibreCAD #578: redundant action prompt updates

r-a-v-a-s, 2015-09-23:
I'm exploring the idea of having the action prompt (left of the command line input)
displayed as command line output.

```c++
void QG_CommandWidget::setCommand(const QString& cmd) {
    if (cmd!="") {
        teHistory->append(cmd);
    } else {
        // lCommand->setText(tr("Command:"));
    }
    leCommand->setText("");
}
```

This reveals redundant updates.
I was able to remove one by commenting out `updateMouseButtonHints` :
```c++
void RS_ActionInterface::resume() {
    updateMouseCursor();
    updateToolBar();
    // updateMouseButtonHints();
    RS_Snapper::resume();
}
```
which follows from
```c++
void RS_EventHandler::mouseEnterEvent() {

    if(hasAction()){
         currentActions.last()->resume();
    } else {
        if (defaultAction) {
            defaultAction->resume();
        }
    }
}
```

However I have been unsuccessful in determining why drawing actions produce two on trigger,
and why panning produces

```
Click and drag to pan zoom
Click and drag to pan zoom
Click and drag to pan zoom
Click and drag to pan zoom
Zoom panning
```

r-a-v-a-s, 2015-09-23:
Note that the previous comment was about middle mouse (scroll wheel) panning.
Ctrl + left_click panning produces nothing.

sand1024, 2026-09-26:
outdated now

closed 2026-09-26 by sand1024