LibreCAD/LibreCAD #581: Block inserting with snapping leads to a segfault

Rupicapra-rupicapra, 2015-09-28:
When inserting a block (circle with solid color hatching) with snap on endpoints/middle/intersection/center activated, passing mouse cursor over drawing area causes LC to crash (terminal shows a `Segmentation fault`).
`git bisect` leads me to this commit : 5ff1f190c1ef3996a028eec87f24b018e48870ed but I can't figure where is the matter in all that changes.

dxli, 2015-09-28:
looks like it's a building issue.

```bash
make distclean
qmake -r
make
```

It works after rebuilding

Rupicapra-rupicapra, 2015-09-29:
I still have segfaults after rebuilding.

Rupicapra-rupicapra, 2015-09-29:
Hum… Seems a little more serious.
I have this problem while I want to draw a polyline snapping on intersections : on a complex drawing (lots of layers, entities…), moving mouse into drawing area while snap on endpoints activated just makes LC crash with a segfault.
On a simple drawing (one layer + layer 0, one line drawn), it occurs when adding another line starting on one of the existing entity endpoint -> crash when clicking to place second point (snapfree activated) (file may be downloaded [here](http://www.jmagnin.fr/owncloud/index.php/s/Rn4joxYbjFQYxCT) ).

dxli, 2015-09-29:
it's weird. I can not get the segfault after rebuilding. It did crash for me initially, and I tried to bisect here with the commit split ( https://github.com/dxli/LibreCAD/tree/issue581 )

We can choose to revert my changes, but I feel it's better, if we can locate the problem. Can you do a bisection on the split branch https://github.com/dxli/LibreCAD/tree/issue581 ?

make sure it does crash with that branch first

Rupicapra-rupicapra, 2015-09-30:
It still fails on this branch (without bisection), but I may have more informations.
GDB returned me this information:
  ````
Program received signal SIGSEGV, Segmentation fault.
RS_EntityContainer::ignoredOnModification (this=0x0)
    at lib/engine/rs_entitycontainer.cpp:1796
1796	    switch(rtti()){

````

dxli, 2015-09-30:
yes, please go ahead with bisection.

it does not crash on my machine, making it more weird to me.

r-a-v-a-s, 2015-09-30:
I confirm the crash after rebuilding.

1. new document
2. free snap + snap endpoints
3. polyline
4. click once
5. click twice

...crash

dxli, 2015-09-30:
I can reproduce this one. I will bisect here.

Thanks!

dxli, 2015-09-30:
pushed commit: cac94fe6

I will review my two big commits more

Rupicapra-rupicapra, 2015-10-02:
It works !
Thanks a lot @dxli

closed 2015-10-02 by Rupicapra-rupicapra