LibreCAD/LibreCAD #585: easier access to "select intersected entities"

r-a-v-a-s, 2015-10-13:
In the default selection mode you can window select by:
- click-drag-release
- click-release-move-click-release

It seems like one of these could be "select intersected entities".

lordofbikes, 2015-10-13:
LibreCAD handle this as many other CAD software.
Drag the window from left to right, it's blue then and only fully included entities are selected.
Drag it from right to left, it's green then and everything that is intersected by the window is selected.

r-a-v-a-s, 2015-10-13:
The difference is that "select intersected entities" can select the top and bottom of this parallelogram without selecting the other sides.
[image]

r-a-v-a-s, 2016-09-07:
Custom toolbars and menus are sufficient.

closed 2016-09-07 by r-a-v-a-s