LibreCAD/LibreCAD #587: How to get perimeter, area of files for .pdf, .dxf, .svg files?

nikhilbhansali, 2015-10-13:
Hey
This is for cost calculation of laser cutting and CNC routing, is there a way to get perimeter? 
Please do let me know!

lordofbikes, 2015-10-13:
In the status bar is a box where you could see the number of selected entities and their total length. Select all entities belonging to the tool path and find the total length (perimeter) in this box.
The entities must not be connected.

To measure an area use the Info->Polygonal Area tool. Click multiple points on the perimeter of the tool path and find the Circumference and the enclosed Area printed in the Command line history.

dxli, 2015-10-14:
@nikhilbhansali 

To get total length, just select entities of a contour, on the status bar there's an area showing selected:

number of entities selected
Total length

To get total area enclosed, you can use the "Hatch" feature. Select the contour, create hatch for it, and total area will be shown in command widget:
[image]

dxli, 2015-10-14:
@nikhilbhansali 

The total length and area features support: line, arc, and elliptical arc types.

closed 2015-10-17 by dxli