LibreCAD/LibreCAD #592: Print scale changes when re-opening file

Rupicapra-rupicapra, 2015-10-22:
Hello there,
I see that print scale is not stored (or at least not used) when saving & closing file, even when 'fixed' is checked, so that you have to re-type scale each time you want to re-print your document. Is it a limitation of the file format, or non-implemented code, or a bug ?
Steps to reproduce :
 1. Open / create a dxf with some entities
 2. Print preview -> set scale -> check 'fixed'
 3. Save & quit
 4. Re-open file and go to print preview.
Thanks,
Rupi

r-a-v-a-s, 2015-10-22:
Check out: `QG_PrintPreviewOptions::destroy`
`cbScale` is the combobox.
First thought: see if `updateDisabled` is the problem.

Something else I noticed is that the print preview toolbar icon is not showing the options.

r-a-v-a-s, 2015-10-26:
I fixed the issue of the options not showing;
I guess I'll look into the settings issue since there has been no update.

r-a-v-a-s, 2015-10-26:
The saving seems to work;
however, a foreign value is added to the combobox at startup.
I think the issue has to do with the `f` variable in:

```c++
void QG_PrintPreviewOptions::updateScaleBox(const double& f){
    int i;
    for(i=0;i<cbScale->count();i++){
        QString s=cbScale->itemText(i);
        int i0 = s.indexOf(':');
        bool ok1,ok2;
        double n = s.left(i0).toDouble(&ok1);
        double d = s.mid(i0+1).toDouble(&ok2);
        if(! (ok1 && ok2)|| fabs(d)<RS_TOLERANCE) continue;

        if(fabs(f-n/d)<RS_TOLERANCE) break;
    }
    if(i<cbScale->count()){
        cbScale->setCurrentIndex(i);
        return;
    }
    QString s("");
    if(f>1.){
        s=QString("%1:1").arg(f);
    }else{
        if(fabs(f)>RS_TOLERANCE) s=QString("1:%1").arg(1./f);
    }
    if(cbScale->count()>defaultScales){
        i=defaultScales;
        cbScale->setItemText(defaultScales,s);
    }else{
        cbScale->addItem(s);
        i=cbScale->count()-1;
    }
    cbScale->setCurrentIndex(i);
}
```

Maybe rallaz or dxli can work on this.

r-a-v-a-s, 2015-11-10:
I did some more testing on this last night and found that `updateScaleBox(const double& f)` is being called three times the first time you activate print preview; and `setCurrentIndex` is called twice toward the top of the function and then once at the bottom. It seems like there is a major flaw in the logic somewhere.

r-a-v-a-s, 2015-11-10:
The associated functions:
QC_ApplicationWindow::slotFilePrintPreview
QG_DialogFactory::requestPrintPreviewOptions
QG_PrintPreviewOptions::setAction
QG_PrintPreviewOptions::scale

Cantar4, 2018-03-07:
Not a show stopper but quite unconfortable ; it leaves the impression that LC house is abandonned to cold winds.

PR #1050 (2019-01-05): Respect paper scale with 'fixed' checked. Issues #592, #605, #705.
This pull request attempts to fix #592, #605, #705.
Also it tries to make paper always fully visible when switching to preview view thus making it more consistent with any drawings.

### Test cases I checked

#### Test drawings

All drawings have paper size A3, landscape.

1. Paper scale is not set, drawn a rectangle which is less than paper.
2. Paper scale is not set, drawn a rectangle which is bigger than paper.
3. Paper scale is 1:10 saved to drawing, drawn a rectangle which is less than paper.
4. Paper scale is 1:10 saved t [... cut]
closed 2019-01-05 by lordofbikes