LibreCAD/LibreCAD #593: more efficient debugging

r-a-v-a-s, 2015-10-28:
We are currently compiling almost 2000 `RS_DEBUG->print(...)` calls and their strings in release mode.
Even though the messages are not printed, mouse movements in the drawing area produce hundreds of needless function calls and expression evaluations in only a few seconds.

The proper approach seems to be to use qDebug() and qWarning().
>qDebug(), qInfo(), and qWarning() are debugging tools. They can be compiled away by defining QT_NO_DEBUG_OUTPUT, QT_NO_INFO_OUTPUT, or QT_NO_WARNING_OUTPUT during compilation.

http://doc.qt.io/qt-5/debug.html

qInfo() is 5.0 + I think.

Then in the pro file you use something like:
```
CONFIG(debug, debug|release) {
  message("Debug")
  DESTDIR = $$DESTDIR-debug
  CONFIG += debug
  DEFINES += DEBUG
  TARGET = $$TARGET-debug
}else{
  message("Release")  
  DEFINES += QT_NO_DEBUG_OUTPUT
  DESTDIR = $$DESTDIR-release
  TARGET = $$TARGET-release
}
```
(not my code)

In theory I could use regex replace to convert the whole project;
however, I'm going to start with some more optimization of the mouse movement code.

lordofbikes, 2015-10-29:
Please, don't do that, there is nothing wrong with RS_DEBUG Class.

Not every user can build LC from source to get a debug version if things go wrong. With RS_DEBUG class it's possible for us to get a user to start his LC release executable on a higher debug level and submit the output to us for debugging. We must not git checkout to an older release and build a debug version to recognize that we still can't reproduce the issue.
That was already helpful in past, when errors were not reproducible in a Qt debugging session or on a different OS.

There is also nothing wrong with QDebug classes and QT_NO_DEBUG_OUTPUT, I use them too. But their approach is not the goal of RS_DEBUG class. Practically RS_DEBUG it as logging class, only the naming failed.

Maybe there is space for optimization, by reducing the number of messages or checking the appropriate levels. But with todays computing performance, I don't see a major loss in LC performance  by RS_DEBUG.

r-a-v-a-s, 2015-10-29:
> it's possible for us to get a user to start his LC release executable on a higher debug level and submit the output to us

Ahh I understand.

I'll be able to eliminate a few messages without touching the debug calls.

A performance gain wasn't my goal with the mouse movement optimization.
It's just inefficient: 
thousands of computers producing thousands of unnecessary calls and evaluations
quickly adds up to millions...
It's like paying mandatory insurance that will probably never do anything for you.
It also quickly pollutes the output for anyone who wants to read it,
and doesn't care about mouse movements.

I think the proper practice is to have the mouse movement code be as clean as possible.
For debugging it could help to make a flow chart or some other description of the code.
Developers can then review that and uncomment the debug calls if needed;
or we could use qDebug and the .pro modification just for the mouse movement.

closed 2016-09-25 by r-a-v-a-s