LibreCAD/LibreCAD #595: CAD Toolbar isn't drawing correctly on a 4K monitor at larger window heights

hansomli, 2015-10-31:
LibreCAD 2.0.4 with maximum window height on a 4K monitor:

[image]

LibreCad 2.0.8 with maximum window height on a 4K monitor:

[image]

The issue goes away with window height this size and below:

[image]

Reproduced on Ubuntu 14.04.3 LTS, 15.04, and 15.10 with librecad from universe and from the stable ppa.

r-a-v-a-s, 2015-10-31:
Thanks for the screenshots; they are always helpful. :sunglasses: 

It is known that the cad toolbar has issues in general.
It's forced and overly complicated.
The QToolBar is designed for one column (or row).

I have created various other replacement widgets,
and I still have more ideas to explore.
http://forum.librecad.org/GUI-update-td5712246.html

r-a-v-a-s, 2015-10-31:
Also note there is an icon size option now... :-]
https://github.com/LibreCAD/LibreCAD/issues/543

dxli, 2015-10-31:
@r-a-v-a-s 

is the "icon size" option  icon itself large enough to be useful on HD?

r-a-v-a-s, 2015-10-31:
It depends how large an icon looks at 150x150 on a 4k screen.
Otherwise we will need to convert to SVG (or make larger icons).
I noticed we have a folder with some SVG icons already.
https://github.com/LibreCAD/LibreCAD/tree/master/desktop/graphics_program_icons

r-a-v-a-s, 2016-02-14:
There's really nothing we can do about this;
and the CAD toolbar has been replaced.

closed 2016-02-14 by r-a-v-a-s