LibreCAD/LibreCAD #596: getting stuck in the plug-in event loop

r-a-v-a-s, 2015-11-01:
I fixed one case where you activate a plugin, then press Esc, then exit the program. https://github.com/LibreCAD/LibreCAD/commit/e3771d16682c44608165563e6bfbd27bef10ad05

However, there is still the case where you activate a plugin and then exit the program.

The event loop can be seen in `Doc_plugin_interface::getSelect`.

r-a-v-a-s, 2015-11-01:
Actually the Esc case still crashes if you don't select an entity before pressing Esc (and then exit).

r-a-v-a-s, 2016-02-05:
The issue is that we currently have three functions in Doc_plugin_interface 
(getPoint getSelect and getEnt) that enter a while loop to wait for the user to do something.

We need to be able to exit this loop in several cases:
- right-click
- Escape
- switch to another tool
- exit the program

I think I solved the right-click case (see previous commit);
however, I didn't have any luck with the others.

My first thought was to get rid of the while loop completely;
however, I have not found a way... maybe if I study the plugin system more.

I thought the Escape case would be simple;
however, qc_actiongetent doesn't seem to receive the keyPressEvent.
I discovered that QC_ApplicationWindow::keyPressEvent is still receiving;
however, removing that didn't change anything.

I thought I have found a solution for everything by changing
`while (!a->isCompleted())`
to 
`while (!a->isFinished())`
however, this crashes the program... and I can't figure out why.

I thought I might be able to solve the program exit case by
replacing the loop body with:
```
    qApp->processEvents();
    if (qApp->closingDown())
        break;
```
        
however, this didn't seem to change anything...

So what happens is that unless you complete the plugin action
or right click to exit it, the program continues inside of one of these while loops;
and when the program is exited by the user, it appears to close;
however, the process is still running in the background and must be forced to terminate.
This is a rather serious issue and I'm afraid unless we solve it we will need to remove
the plugins that use these loops, or try to convert them to core code tools.

I'll be moving on to other issues for now,
and encourage others to try and solve this.

r-a-v-a-s, 2016-02-05:
Of course after writing all that... I think I solved everything. ;-]
Let me know if I missed anything.

closed 2016-02-12 by r-a-v-a-s

Xia-Weiwen, 2016-09-12:
@r-a-v-a-s Would you please try the plugin called Same Properties? It seems to cause a crash.
I'm using Ubuntu 16.04 LTS 64 bit with Unity.
Following these steps may cause crash:
1. Draw two lines.
2. Activate the Same Properties plugin.
3. Select one line by mouse left click.
4. Select the other line and press enter.
5. If the program doesn't crash, try 1-4 again.
The problem seems to be in the ```Document_Interface::getEnt```. It will return a invalid pointer (not NULL) sometimes.
I've tried to write a sample plugin which only contains ```Document_Interface::getEnt``` and ```Plug_Entity::getData```. The program will crash when excecuting ```Plug_Entity::getData``` because the Plug_Entity pointer returned by ```Document_Interface::getEnt``` is invalid (not NULL, the pointer points to 0xc0 when debugging).
The wierd thing is that sometimes it works and sometimes it will crash.
If you find something, please tell me. Thank you!

r-a-v-a-s, 2016-09-12:
@Xia-Weiwen This issue is about getting stuck in a loop;
please start a new issue.