LibreCAD/LibreCAD #597: segfault by grid options in the master branch

dxli, 2015-11-01:
steps to reproduce:

0, start LC;
1, Main menu: Settings -> Current Drawing Options -> Grid;
2, switch between orthogonal/isometric grid types. For example, if the current grid type is orthogonal, check the "isometric" radio box.

Results:

LC crashed.

The first commit introducing this crash is 84fd057b1f7cafa4d782d0b02ff2f265c5f8d05f

and this bug is by the line of code in [qg_dlgoptionsdrawing.cpp](https://github.com/LibreCAD/LibreCAD/blob/master/librecad/src/ui/forms/qg_dlgoptionsdrawing.cpp) by the line:
```C++
			graphic->addVariable("$DIMTXSTY", cbDimTxSty->getFont()->getFileName() , 2);
```

due to invalid cbDimTxSty->getFont() pointer

fa201, 2020-04-14:
@lordofbikes 
I cannot reproduce this bug.
Version: 2.2.0-rc1-190-g51b21cc7
Compiler: GNU GCC 7.5.0
Compiled on: Mar 29 2020
Qt Version: 5.9.5
Boost Version: 1.65.1
System: Ubuntu 18.04.4 LTS

sand1024, 2026-09-26:
not reproduceable, too old

closed 2026-09-26 by sand1024