LibreCAD/LibreCAD #598: merging modify → properties & attributes

r-a-v-a-s, 2015-11-08:
As a new user the difference between the properties and attributes tools is confusing.
Adding to the confusion: selecting objects is different for the two.

I imagine a single action that is not checkable (no toggle).
You trigger it after you select one or more object(s).

- one selected object → properties window
- two selected objects of the same type → modified properties window
- two selected objects of a different type → attributes window

In the case "two selected objects of the same type":
we will want to be able to set each property individually;
therefore, the dialog would benefit from becoming non-blocking,
which would also allow for an apply button.
The other option is to make it a QDockWidget.

kfoltman, 2016-03-03:
I think previously it was possible to set the same property value for multiple objects using Properties, now it's not possible, as the command ignores the current selection. It's often a useful functionality, for things like setting radius on multiple circles.

A dockable dialog seems like a good idea, too, but there are some subtle problems, like - what happens to selection after applying? If it stays, then multiple objects can be inadvertently modified. If it doesn't, then doing a sequence of operations on a selection (modify radius, then move) might get harder.

A dock widget is a good idea. DraftSight uses something similar - a property sheet.

r-a-v-a-s, 2016-03-03:
I would be inclined toward letting the selection remain unchanged.

@vlachoudis might already be working on this feature.

Modiug, 2024-12-30:
OK, this is rather old.
Coming from QCad, I was used to select multiple circles of the same type (e.g. I organized them in the same layer) and change the radius for all of them in Properties. That indeed only seems to work for one item now (to be selected by clicking on it).

Has this been fixed or is this functionality available in a different way?

ConnerMcLaughlin, 2025-01-19:
As far as I have discovered, there is no current way for LibreCAD to batch modify "geometry" properties for multiple entities.  The closest I have got was using the Scale tool after selecting more than one entity.  The side effect of simultaneously scaling the spacing between the entities is a deal breaker for many projects.  Maybe some kind of extra option in the Scale tool that separates the application of scaling to the entities sizes rather than their spacing or vise versa.  Also another option that would help is to have the ability to use absolute scaling base on the units chosen for the current drawing preferences.

sand1024, 2026-09-26:
supported by Properties widget in 2.2.2

closed 2026-09-26 by sand1024