LibreCAD/LibreCAD #601: custom calculator constants (per-drawing, if possible)

Osndok, 2015-11-09:
I noticed somewhere that one could have 'pi' in most fields, and that it would behave well.

I think that it could be quite a timesaver to have user-definable per-drawing constants; as then a power user might just enter "a", "b", or "c" into (say) a length field rather than having to lookup elsewhere "1.1581".

dxli, 2015-11-11:
sounds reasonable.

somewhere in preferences a list of constants to use.

Please consider helping the project with the coding part.

r-a-v-a-s, 2015-11-12:
I would avoid QSettings and allow for people to select their own .txt file:

```
T = 6.2831853071795865
g = 0.6180339887498948
...
```

This requires less coding,
and intrinsically produces a back-up file.

Osndok, 2015-11-12:
I might be able to provide a work plan, but I don't know how much of this I could do myself:

* submerge degree-to-radian logic (found in RS_EventHandler::cliCalculator) into RS_Math::eval (or into a new RS_Math wrapper function)

* add RS_Math::assignConstant(identifier, expression) : expression is evaluated immediately [for now], and keyed to identifier

* add RS_Math::clearConstants() to eradicate all constants (and re-add 'pi', if needed)

* RS_Math::eval() uses assigned constants in generating results

* add RS_Commands::filterAssignment (like filterCliCal) to detect/extract (identifier, expression) tuple. '\s*([^0-9][a-zA-Z0-9]*)\s*=(.*)' ???

* add RS_EventHandler::cliAssignment (much like cliCalculator) to handle assignments from command box

* at this point, user-defined constants should work, but are not persisted or reloaded; a good point for some careful testing

* add application preferences (under 'paths') 'Math Constants'

* if the file exists after pref value change or at startup
** RS_Math::clearConstants()
** for each line of the file
*** Skip empty lines, and those starting with '#'
*** (identifier,expression) = RS_Commands::filterAssignment(line)
*** RS_Math::assignConstant(identifier, expression)