LibreCAD/LibreCAD #602: Enable hatching using flood-fill and autodetect/autocorrect problems with hatching

Harvie, 2015-11-11:
Hello,
i've tried to hatch some of more complex drawings and it's pain. I have everything in separate layer and still cannot do it. It alwas says that the contour is not manifold and there is somthing like points 321.000,256.000 and 321.001,255.999 are not touching. (Even on drawing where i set 0.0 precision!) However even if i delete the lines and make it again using "snap on ends" so it should snap tightly it still does not work, or just unsnaps on the other end of the line.

It would help if you treat points so close to each other as one.

Also may help if there was way to automatically detect and replace duplicate/overlapping lines in drawing to make sure there are no invisible lines under another (in each layer separately). QCAD have some "detect duplicates" feature that i haven't seen yet, but maybe you can just backport it.

Having flood-fill hatching would be great... Maybe you can do it easier on bitmap level, export drawing to very high resolution bitmap, fill it with flood hatches (around some defined "flood inlet" points) and then try to trace the bitmap hatch back to vectors.

BTW i am using LCAD 2.0.8

dxli, 2015-11-11:
hi Harvie,

hatch is definitely the weak point now.

We mentioned:

1, allow contour by intersections;
2, efficient flood-fill.

It's very interesting algorithm work. Would like to help the actual development side?

Thanks,

dxli

Harvie, 2015-11-11:
Is there any plugin API? Like something that will enable me to write simple script and then call it from menu like tool? Good starting point would be some interpreter with API to access the drawing entities (i don't know how DXF works, i suppose it's bunch of lines, points and splines or something like that). Also would be great if that plugin API could pass info about what entities are selected, which layer is active, etc...

So people can write simple plugins. FreeCAD does this and there are sh*tloads of 3rd party plugins some of which are being merged to main repo as official tools. It's cool idea to decentralize the development even more.

Also enables people to do some simple task by writing simple one-shot code rather than drawing entire thing manualy. Let's say i'll want to draw l-system tree. It's much easier if i can write 8 lines long script using some plugin API and execute it in context of my drawing (possibly with some parameters).

dxli, 2015-11-12:
Hi @Harvie ,

Actually, there is a plugin interface. Please have a look at the current plugin examples:

https://github.com/LibreCAD/LibreCAD/tree/master/plugins

and the interface for plugins:

https://github.com/LibreCAD/LibreCAD/tree/master/librecad/src/plugins

Please feel free to play, ask for help and send in pull requests.

Harvie, 2015-11-12:
That's not exactly what i've been thinking of. It's good to have binary plugins, but it would also be interesting to have framework for interpreted scripts which one can write and execute directly in librecad interface. That's a bit different thing, but it may provide another new level of end-user functionality. However this ticket is about hatches, not plugins.

I haven't seen the librecad code yet and i am quite busy right now, maybe someday i will have more time, but now i have to be drawing in CAD, not programming the CAD :-)))

dxli, 2015-11-12:
@Harvie 

LibreCAD v3 aims to support real scripting, but needs developers.

lordofbikes, 2018-05-24:
See http://forum.librecad.org/Hatching-always-failed-working-with-ellipse-and-line-tp5716011.html too