LibreCAD/LibreCAD #604: Git integration?

Harvie, 2015-11-11:
Just idea... But what if i can commit and push my drawings through librecad? Some people may also want to setup automatic commit to be done when one saves the file.

dxli, 2015-11-12:
@Harvie 

First, some tricky part:

1, dxf seems to be not really human readable;
2, current dxf format uses external images, not included in dxf itself;
3, svg might be a better candidate, need better SVG reader/saver for LibreCAD.

Let's make a git plugin and see whether users like it.

Harvie, 2015-11-12:
Git support should not be dependent on which format you use. You use DXF when you need DXF, you use SVG when you need SVG, you use GIT when you need versioning of whatever you draw. Should be quite easy, when enabled just call git commit "whatever/file_you_just.saved" and optionaly the git push.

I don't understand how external images are problem for this. You can have them in repository too. However using relative paths is more than preferable in such case.

You don't need diffs to be human readable in order to have versioning working. Also there are ways around problems with human readability of diffs. if someone really needs it, then it can always be solved by custom diff filters in git.

dxli, 2015-11-12:
@Harvie 

I mean it's efficient for file format with readable patches.

For plain text, while the diff sets may not be easy to understand, storage should be efficient.

For binary format, git simply works like a backup tool.

Harvie, 2015-11-12:
Backup tool is often all you need when things got messed :-) I have written dokuwiki plugin that can edit SVGs using javascript editor and stores them as plaintext in dokuwiki's versioning backend. What i've found is that to some extent the diffs can make sense, however in real life they are still pretty useless :-) You see some entities/lines added or removed, but more important is what these lines mean.

You will not understand from diff that somebody drawn a car park where you wanted to have garden. That's why git has commit comments. So librecad can ask you for commit comment before every commit

And once again - there are programs that can do diffs between images and git supports custom settings on how diffs are generated for various filetypes. You can even instruct git to transparently convert DXFs to temporary SVGs (or something else) before doing diff. So you work with DXF, but in git log you see diff of SVGs. It works suprisingly well. So this is absolutely nothing you should be concerned about.

r-a-v-a-s, 2015-11-12:
I like that no part of the code tries to access the internet.

We already have a lot of options for pushing: 
http://git-scm.com/downloads/guis

kromanenko, 2015-11-12:
I have had thoughts to use git for sharing AutoCAD models, but really it`s weird way. Git is good for code, not for binary data, nevertheless it is also possible.

DuncanLithgow, 2021-04-03:
you guys might be interested in http://openingdesign.com/about/ an architectural practice who do all their design work via github.

fa201, 2023-08-08:
Is this something that we want to push ?

Basically, commiting with whatever git-solution is easy and quick. I mean, I do not see a significant gain in doing this in LibreCAD vs doing in a git-solution (apart from selecting the drawing which would be automatic in LibreCAD).

Any one has thoughts on this ?

lordofbikes, 2023-08-09:
Sure, and that's why I close it now 😉

There is many true of what was discussed yet, but also some wide-eyed suggestions.

Providing a versioning resp. backup solution, includes a handy restore solution too.
Even just committing locally will not be an easy-peasy implementation.
We would need at least the `log` command, to list revisions of a file and `checkout` to restore a previous version. But what happens, when an in between version is modified and committed?
And how to handle merge conflicts, deleted files or what else I can't imagine yet?

So in the end we would need to implement a complex `git` GUI inside LibreCAD.
For users, who anyhow know the `git` workflow and probably already have a well working GUI this might not be a satisfying solution. And for the `git` 'foreigners', I see more issues arising when something goes wrong with the repo.

Also, in more than 7 years, there was nobody taking the challenge yet.

There exist more than one good `git` GUI and also online git platforms, all well documented and supported and many for free. So why reinvent the wheel?

closed 2023-08-09 by lordofbikes