LibreCAD/LibreCAD #606: Macro to print scale in text

Harvie, 2015-11-11:
I think that it will be really usefull if there were some macros that will be processed before printing.

Like if i add some text to drawing and it will contain some strings like "$SCALE$" or "$UNITS$" it will enable me to have correct value printed on paper.

Now if i add text eg. "This drawing is made in scale 1:50" and then decide to print it on A3 instead of A4 it will not change automatically. So i think it would be cool to say just "This drawing is made in scale $SCALE$" and it would be captioned properly in any scale.

BTW also tabs are not rendered correctly in text entity for some reason.

dxli, 2015-11-12:
@Harvie 

in addition, some big ones we have talked:

1, native TTF support, basically, using fonts available for Qt, instead of the ones shipped with LibreCAD. The goal, a whole page of text should be light to process;
2, better language support (like from right to left languages).

Harvie, 2015-11-12:
step one: make "unicode font" the default for mtext.

Also i am not sure if TTF is that great idea. Doesn't it somehow interfere with compatibility? What if i send my drawing to someone who will open it in AutoCAD on Windows with completely different set of fonds available? Would be cool if DXF had some way to store text both like lines and ascii, so drawing would be more portable, but text will be still editable.

dxli, 2015-12-04:
Somewhat related.

Support for "Right to left" languages, for example, Hebrew and Arabic.

I think it's a little bit more complex than placing alphabets from right to left.

Harvie, 2015-12-04:
@dxli any particular reason why you don't open separate ticket for this "somewhat related" (but completely different) issue?