LibreCAD/LibreCAD #612: touchscreens and gestures

r-a-v-a-s, 2015-11-14:
I have a touchscreen monitor now;
therefore I will be working on enhancing the touchscreen experience.

I will start with two finger panning and zooming.

I noticed someone added:
```c++
bool QG_GraphicView::event(QEvent *event) {
#if QT_VERSION >= 0x050200
    if (event->type() == QEvent::NativeGesture) {
        QNativeGestureEvent *nge = static_cast<QNativeGestureEvent *>(event);

        if (nge->gestureType() == Qt::ZoomNativeGesture) {
            double v = nge->value();
            RS2::ZoomDirection direction;
            double factor;

            if (v < 0) {
                direction = RS2::Out;
                factor = 1-v;
            } else {
                direction = RS2::In;
                factor = 1+v;
            }

            // It seems the NativeGestureEvent::pos() incorrectly reports global coordinates
            QPoint g = mapFromGlobal(nge->globalPos());
            RS_Vector mouse = toGraph(RS_Vector(g.x(), g.y()));
            setCurrentAction(new RS_ActionZoomIn(*container, *this, direction,
												 RS2::Both, &mouse, factor));
        }

        return true;
    }
#endif
	return QWidget::event(event);
}
```

This does nothing on Windows 7.
The [docs](http://doc.qt.io/qt-5/qnativegestureevent.html) for `Qt::ZoomNativeGesture` say:
"OS X: Two-finger pinch." 

http://doc.qt.io/qt-5/qpinchgesture.html
has existed since 4.6 and doesn't seem to be OS specific.

http://doc.qt.io/qt-5/gestures-overview.html
http://doc.qt.io/qt-5/qtouchevent.html

rubdos, 2016-06-18:
These `QGesture`s don't seem to work when using XWayland (i.e., the libinput events don't get forwarded through XWayland or [this bug?](https://bugreports.qt.io/browse/QTBUG-14224)). I didn't try using X.org itself yet, but I'd like this issue to be future proof (and thus, Wayland proof).

r-a-v-a-s, 2016-06-20:
What are you hoping for exactly?

It's been a while since I worked on this,
but I did get the pinch/zoom and two finger pan working on Windows 7.
The problem is that they conflict with one another,
and I didn't find an easy solution.

Another issue is parallel synthesized mouse events;
I was thinking about just ignoring them
and writing duplicate event handling for touch events.
But then I realized mouse is much more practical than using a touchscreen,
and that Qt's touchscreen support isn't fully developed yet.
So I lost most of my motivation for all this.

Another issue is LibreCAD's default selection mode.
I was thinking about making a modified default mode for touch devices.

I also found that a two finger tap was useful as right-click for touchscreen,
so I will probably still implement that.
Although it's more like, two fingers down, and then it triggers when the first one comes up.

I'll show you what I've got already in a day or two.

rubdos, 2016-06-21:
>  and that Qt's touchscreen support isn't fully developed yet.

That's what I noticed too. I think it's best to ignore my comment from three days ago, I thought this was about trackpads too.

fa201, 2023-08-09:
Is there a plan or a way for this feature request ?