LibreCAD/LibreCAD #613: Deleting block causes current drawing to close

Rupicapra-rupicapra, 2015-11-16:
Hello there,
When I want to delete a block from the block list, current drawing closes, and then I have an empty LibreCAD window.
Debug message does not show any information, and I can't bisect to find where this comes from because some commits brought compilation errors (solved later).
Thanks for any help,
Jerry

Version : master, updated and built today
OS : Linux Mint 17.2
Qt 4

r-a-v-a-s, 2015-11-16:
I wasn't aware that a window was closed on block removal.
The code needed a small tweak because I eliminated the unnecessary QMainWindow
when I switched QC_MDIWindow to be a subclass of QMdiSubWindow.

closed 2015-11-16 by r-a-v-a-s

Rupicapra-rupicapra, 2015-11-17:
Your bugfix works great. Thanks !