LibreCAD/LibreCAD #614: new preference/feature to default to first settings field (rather than command box)

Osndok, 2015-11-17:
By defaulting to the command box, we are presenting optimizing for cases where the user already has a list of coordinates at hand.

e.g.
1. move mouse & click on 'horizontal line'
2. user types coordinates "0,0<enter>"

However useful (e.g. for transcribing pre-existing drawings), I'm not sure this is even the most common workflow. To me it seems like a special case. At least for my workflow, it would be more useful to optimize the default field focus to allow this:

1. move mouse & click on 'horizontal line'
2. user types length "12.34", possibly "tabbing" to the next settings field (like angle)
3. move mouse & click where the line should be placed

Which is to say, "when an action is clicked, focus on and select all the text of the first settings field".

dxli, 2015-11-19:
@Osndok 

somewhat related.

Like the "Line by angle", there are two fields in option toolbars, "angle" and "length".

I feel most first time users are not even aware of them.

Osndok, 2015-11-19:
That is *very* related, and *very* believable... so the issue (as I have phrased it) is really geared towards people who "already get it"... to the new user, they are surly a bit lost & baffled (as I was, before I watched a youtube tutorial).

I think that there are several parts to the general problem:
1. the common case is that a user would both select a tool & tweak it's options (as I originally said),
2. the option fields are far away from the widget they modify/control,
3. the relevant option fields are visible only after the user selects a tool, and
4. at the time of selecting a tool, the user's focus is 'far' from the just-now-appearing option fields

To be practical, and not abandon the user base, we might address only the most effective of these issues.

However... in theory, we could address all of these... but it would be a lot of work, the interface would look a lot different, and probably very foreign to existing users. I will attach a mockup for discussion...

[image]

Osndok, 2015-11-19:
I would add... the 'restriction' tool group, might be above the guide too.

dxli, 2015-11-19:
looks like the "options" fields can be part of "static property editor" as well.

Two conceptions might be combined, when drawing a line, user can expect "length" and "angle", but can also use the fields to edit the properties of existing lines.

Osndok, 2015-11-19:
I think that you are right. Even if they are not technically the same widget, as long as the layout of the two are **similar** and in the same **location**, it would go a long way towards usability.

Osndok, 2015-11-19:
Another mockup, this one is the angle detail showing two complements (180 & 90), and measure.
[image]

r-a-v-a-s, 2015-11-19:
Currently the tabs are created automatically. You can drag and drop any of the dockwidgets on top of another and it "tabbifys" them. You could definitely make another dock-widget that contains all of this if you want. The only thing I'm opposed to is side tabs, because the text is hard to read and seems out of place. An alternative idea is to use the category icons as QToolButtons to control a QStackedWidget.

Widgets to consider:
QDockWidget
QTabWidget
QStackedWidget controlled by QComboBox or QToolButton(s)
QWidget, QFrame, QGroupBox, QScrollArea
QLabel, QLineEdit, QSpinBox, QCheckBox

http://doc.qt.io/qt-5/layout.html

r-a-v-a-s, 2015-11-19:
Also look in `librecad/src/ui/forms` at files like `qg_lineangleoptions`.

r-a-v-a-s, 2015-11-20:
My plan was to make a dockwidget alternative to the toolbar.
That way the options remain unbound to the tool icons,
which allows for more user customization.
However, it keeps getting knocked down on my priority list
(my focus is geometric construction which doesn't use tools with options).