LibreCAD/LibreCAD #616: Plugin "List Entities" crash

dxli, 2015-11-26:
Affected version: 2.0 (expected in master branch, but not tested yet)

steps to reproduce:

* start LC;
* Draw -> Line ->Rectangle;
* select first and and second corner to draw a rectangle;
* Info -> List Entities;
* when proposed to select entities, select the rectangle drawn;
* Click "Continue" of the selection toolbar.

Expected results:

* entities listed;

Actual results:

* LC crashes with dereferencing of nullptr

Reproducibility: 100%

Rallaz, 2015-11-26:
Tested with LibreCad 2.0 +git 2015-10-19 & 2.0 +git 2015-11-26
Made the steps, actual result : 
Entity are listed without crash.
Maybe your LC found a incompatible liblist.so file?

r-a-v-a-s, 2015-11-26:
See https://github.com/LibreCAD/LibreCAD/commit/e3771d16682c44608165563e6bfbd27bef10ad05

dxli, 2015-11-26:
@Rallaz @r-a-v-a-s 

sorry, my bad. After rebuilding, 2.0 does not crash

master's branch still crashes for me, by the dtor of ActionGetSelect

r-a-v-a-s, 2015-11-26:
https://github.com/LibreCAD/LibreCAD/issues/596

r-a-v-a-s, 2015-11-26:
What steps in the master branch? There is no continue button.

dxli, 2015-11-26:
@r-a-v-a-s 

Just press Enter instead of the "Continue" button

r-a-v-a-s, 2015-11-26:
It works fine for me.

closed 2015-11-29 by dxli