LibreCAD/LibreCAD #622: feature request: properties as dockable dialog

vlachoudis, 2015-12-14:
I've recently was experimenting with Autodesk fusion 360 and I liked the feature that you can add dimensions on entities that allow you to edit an modify it. e.g. add a dimension to a line and then resize the line to size you want.

I understand that maybe the dxf file format doesn't allow to store these information, but what if
you could provide the properties dialog as a dockable panel and every time we select an entity to see its values and possible change them. Also to provide further "derived" properties like the length of a line, angle, center point, etc...

Even more advanced would be able to give the possibility to insert a function or a variable/parameter there so to allow some parametric drawing.

SmartsOnSite, 2016-08-31:
I recently loaded a DXF drawing that was created using MasterCAM, the lines are all green and I want to inherit their colour from the layer. To do this I had to select each line, click on the colour combo-box, select "By Layer", then click OK, then select the next line, select the colour combo box, select "By Layer" then select... you see the problem. A way to modify common properties of selected items would be SO useful.

r-a-v-a-s, 2016-08-31:
@SmartsOnSite Modify->Attributes

SmartsOnSite, 2016-08-31:
Hi r-a-v-a-s, thanks for that. I was looking for "properties" or a context popup menu on the right mouse button. That works a treat.

sand1024, 2026-09-26:
supported by properties widget in 2.2.2

closed 2026-09-26 by sand1024