LibreCAD/LibreCAD #629: Status bar display error

nikonboy, 2016-01-06:
Hello

The status bar at the bottom does not render properly on a surface pro 3, making the software unusable.

nikonboy, 2016-01-06:
[image]

r-a-v-a-s, 2016-01-06:
I've noticed that removing `statusBar()->setMinimumHeight(32);` allows for proper automatic sizing;
however, it makes the statusbar height dynamic, which is noticeable when a status tip is displayed. 

I'll be adding an option for the height soon.

r-a-v-a-s, 2016-01-08:
You can find the statusbar height option in "Widget Options".
https://sourceforge.net/projects/librecad/files/Windows/NightlyBuilds/master/

closed 2016-01-15 by r-a-v-a-s