LibreCAD/LibreCAD #630: User Manual

tytower, 2016-01-09:
Great to have writers prepared to do the hard yards making this up .
One criticism though , It becomes tedious and boring and distracting if you don't just deal with the subject . Writers of this manual seem to be diverting off into side issues and repeating the obvious to try to make it user friendly I suppose.

Here is an example "Now we are going to touch on Layers. These are especially important in certain drawings. I want to point out one example now and I am going to get kind of technical here so please bear with me. Engineers will understand what I am saying but this may be more challenging to understand for others so I am going to try to make it as simple as I can. Layers are especially important in Assembly drawings."

This under the heading "Layers"
Now all of this is superfluous rubbish and totally unnecessary. 

This seems to be the case all the way through . If you are dealing with "Layers" say
Layers
Layers are used to separate parts of the drawing into manageable chunks.............etc

Get right into it straight away dont procrastinateplease . That Manual could be cut by two thirds

dxli, 2016-01-10:
@tytower 

we may also want to include a copy of rst user manual in our codebase.

lots of work, clearly.

tytower, 2016-01-10:
LibreCad has  given me a lot of enjoyment learning to use the program . With Bob's consent I would be willing to start at hacking out bits I feel unnecessary to streamline it for inclusion in the code if asked.

dxli, 2016-01-10:
@tytower 

The user manual was originally contributed by Bob Woltz in 2012. Big thanks to him!

Bob was kind enough to allow Creative Common Share Alike License for this doc.

On wiki, this user manual received lots of contributions from many users.

I think it's straight enough for us to keep creative common license for the RST part in code base.

In short, you are free to create your version based on the current user manual on wiki.librecad.org, and Bob already allowed/encouraged this type of work.

r-a-v-a-s, 2016-01-15:
Please post topics related to documentation on the forum.
http://forum.librecad.org/

GitHub issues are things we can close by committing to this repository (changing the program).

closed 2016-01-15 by r-a-v-a-s