LibreCAD/LibreCAD #633: Bug: Core dump after undo

vlachoudis, 2016-01-25:
I am getting a core dump after undoing offset lines and try to create new ones
I am using Fedora Core 23 with librecad 2.0.8.
Steps to produce
1. create a horizontal line
2. select the parallel tool and create two lines one above and one below
3. undo both parallel lines (twice Ctrl-Z)
4. click to create a new parallel line above the horizontal one
```
(gdb) bt
#0  0x000000000057a410 in  ()
#1  0x00000000004e2dae in  ()
#2  0x000000000043585a in  ()
#3  0x00000000005c19d7 in  ()
#4  0x000000000052f288 in  ()
#5  0x0000000000533f49 in  ()
#6  0x000000000061dabe in  ()
#7  0x00007fc18d6fa140 in QWidget::event(QEvent*) (this=0x1af71b0, event=0x7ffd8017b460)
    at kernel/qwidget.cpp:8775
#8  0x00007fc18d6a393c in QApplicationPrivate::notify_helper(QObject*, QEvent*) (this=this@entry=0x1585010, receiver=receiver@entry=0x1af71b0, e=e@entry=0x7ffd8017b460) at kernel/qapplication.cpp:4565
#9  0x00007fc18d6aa956 in QApplication::notify(QObject*, QEvent*) (this=<optimized out>, receiver=0x1af71b0, e=0x7ffd8017b460) at kernel/qapplication.cpp:4108
#10 0x00007fc18d1358dd in QCoreApplication::notifyInternal(QObject*, QEvent*) (this=0x7ffd8017bdd0, receiver=receiver@entry=0x1af71b0, event=event@entry=0x7ffd8017b460) at kernel/qcoreapplication.cpp:955
#11 0x00007fc18d6a9f5d in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool) (event=<optimized out>, receiver=<optimized out>)
    at ../../src/corelib/kernel/qcoreapplication.h:231
#12 0x00007fc18d6a9f5d in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool) (receiver=receiver@entry=0x1af71b0, event=event@entry=
    0x7ffd8017b460, alienWidget=alienWidget@entry=0x1af71b0, nativeWidget=nativeWidget@entry=0x1701d80, buttonDown=buttonDown@entry=0x7fc18e1d63c8 <qt_button_down>, lastMouseReceiver=..., spontaneous=true)
    at kernel/qapplication.cpp:3173
#13 0x00007fc18d724862 in QETWidget::translateMouseEvent(_XEvent const*) (this=this@entry=0x1701d80, event=event@entry=0x7ffd8017b7c0) at kernel/qapplication_x11.cpp:4546
#14 0x00007fc18d72322d in QApplication::x11ProcessEvent(_XEvent*) (this=0x7ffd8017bdd0, event=event@entry=0x7ffd8017b7c0) at kernel/qapplication_x11.cpp:3663
#15 0x00007fc18d74c249 in x11EventSourceDispatch(GSource*, GSourceFunc, gpointer) (s=0x15868b0, callback=0x0, user_data=0x0) at kernel/qguieventdispatcher_glib.cpp:148
#16 0x00007fc18aecce3a in g_main_context_dispatch (context=0x1585520) at gmain.c:3154
#17 0x00007fc18aecce3a in g_main_context_dispatch (context=context@entry=0x1585520) at gmain.c:3769
#18 0x00007fc18aecd1d0 in g_main_context_iterate (context=context@entry=0x1585520, block=block@entry=1, dispatch=dispatch@entry=1, self=<optimized out>) at gmain.c:3840
#19 0x00007fc18aecd27c in g_main_context_iteration (context=0x1585520, may_block=may_block@entry=1)
    at gmain.c:3901
#20 0x00007fc18d165e8e in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) (this=0x1585410, flags=...) at kernel/qeventdispatcher_glib.cpp:450
#21 0x00007fc18d74c3e6 in QGuiEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) (this=<optimized out>, flags=...) at kernel/qguieventdispatcher_glib.cpp:207
#22 0x00007fc18d134151 in QEventLoop::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) (this=this@entry=0x7ffd8017bba0, flags=...) at kernel/qeventloop.cpp:149
#23 0x00007fc18d1344c5 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) (this=this@entry=0x7ffd8017bba0, flags=...) at kernel/qeventloop.cpp:204
#24 0x00007fc18d13a069 in QCoreApplication::exec() () at kernel/qcoreapplication.cpp:1227
#25 0x000000000042c0da in  ()
#26 0x00007fc18beea580 in __libc_start_main (main=
    0x42ac50, argc=1, argv=0x7ffd8017bfd8, init=<optimized out>, fini=<optimized out>, rtld_fini=<optimized out>, stack_end=0x7ffd8017bfc8) at libc-start.c:289
#27 0x000000000042f6c9 in  ()
```

r-a-v-a-s, 2016-01-25:
This sounds like something that was fixed already.
I'm not sure if you can update on Fedora, but we are on 2.0.9 now.

vlachoudis, 2016-01-25:
Thanks. I will download and compile the 2.0.9.

dxli, 2016-01-26:
@vlachoudis 

please let us know if you have trouble with undo with 2.0.9 still

thanks!

vlachoudis, 2016-01-26:
I've download then new master and it works ok.
Thanks

closed 2016-01-26 by vlachoudis