LibreCAD/LibreCAD #634: Spline control points

dreappdev, 2016-01-26:
After drawing a spline through points, the controls points can be selected and moved. But once the drawing is saved, they are no longer available for selection, at least not at the same locations. The control points appear to be moved and trying to bring them back to where the user first specified them results in an inappropriate spline curve. The DXF file does allow for the saving of the control points (absolutely needed to redraw the spline) so perhaps they are being "corrupted" or recalculated when the spline is saved? This needs fixed so they are always the same points the user specified rather than some  unguessable point after the drawing is saved.

dxli, 2016-01-28:
@dreappdev 

This is a known problem.

The current implementation tries to fit the new type of splinepoints with existing dxfrw filters.

Feel free to give us a help hand here.

dreappdev, 2016-01-29:
Well, I just downloaded the DXF format document and started browsing through it. I’ll have to go back to my math/calculus/geom books I think to review how a spline is actually calculated. But I was just thinking that if the DXF format itself is not quite specific or flexible enough to do this, then perhaps the user input/selected points for drawing the original spline could be included in “comment” lines within the DXF file. Other CAD programs would ignore the info, but you could pick it out and use it to recreate the original spline control points.
Just a thought. I am not an expert on the DXF format and certainly don’t know how you have implemented it. Yet. I think I’ll download the source code and see what you did. Can you give me some clues as to where to look? I am pretty good at reading and working with a variety of programming languages and a variety of OS platforms, have retired from the software programming profession just a few years ago.

Thanks, DErik

r-a-v-a-s, 2016-01-29:
I'm assuming [dxfreader.cpp][1] and [dxfwriter.cpp][2] are what you want to look at.

[1]:https://github.com/LibreCAD/LibreCAD/blob/master/libraries/libdxfrw/src/intern/dxfreader.cpp
[2]:https://github.com/LibreCAD/LibreCAD/blob/master/libraries/libdxfrw/src/intern/dxfwriter.cpp

r-a-v-a-s, 2016-01-29:
[libdxfrw.cpp](https://github.com/LibreCAD/LibreCAD/blob/master/libraries/libdxfrw/src/libdxfrw.cpp)

dreappdev, 2016-01-29:
> I'm assuming dxfreader.cpp <https://github.com/LibreCAD/LibreCAD/blob/master/libraries/libdxfrw/src/intern/dxfreader.cpp> and dxfwriter.cpp <https://github.com/LibreCAD/LibreCAD/blob/master/libraries/libdxfrw/src/intern/dxfwriter.cpp> are what you want to look at.

Got those (via link in this email, haven’t downloaded the whole source yet). They are merely the “helper” or utility routines that read and write whatever. It’s *what* is written and into what part of the DXF file that is important. Where are the spline control points written. IE, where is the call to dxfwriter that outputs the spline control points; and the subsequent calls to dxfreader that recreates the spline?

Thanks, DErik

r-a-v-a-s, 2016-01-29:
Check `dxfRW::processSpline` and `RS_FilterDXFRW::addSpline`.

dreappdev, 2016-01-29:
Thanks. Got this too. Getting closer. I found the “dxfRW::writeSpline” procedure. Will take a bit more time to review this and correlate with other things; it’s a lot bigger source file. I need to look at other files to determine what the source of the data looks like and how it is derived/calculated. This helps me.

Thanks, DErik

dreappdev, 2016-01-29:
Will do.  Thanks for the quick responses. I take a good hard look at this stuff and hopefully I can help out. If not with actually fixes for you, but perhaps at least helpful and welcome suggestions.

Thanks, DErik

dreappdev, 2016-01-29:
Was just looking at the source code web page you have. Browsed through the “libraries” directory. Looks like you are using Visual Studios (on Windows platform) or Xamarin (on OS platform) to build this. That’s fine, but can I get the whole project downloaded to my laptop? That way I can research it “at my leisure” and play with options without corrupting your baseline. Or is there a serious license problem? I won’t be able to replace any of the files into your baseline, and I promise to adhere to any restrictions you want to impose. Nor will I disclose the source to anyone else unless you authorize it. I just want something better that browsing through the files on-line, like from within the Xamarin IDE, to make the research easier. I’m currently running on Mac OS X 10.11.2 (El Capitan). My Windows platform has seen its last days. I used to develop a few things in C# using VS on the Windows platform. I’ve tried converting that source to Xamarin on OS X but the GUIs are quite different, so I tabled that. But if you could zip up the relevant directories that VS uses, I could sure use that.

Thanks, DErik

dreappdev, 2016-01-29:
BTW, I thought you might be a little interested in why I’m interested in this app. Well, I’m interested in model trains. I’m building a garden railway and am going to build a lot of my own model buildings (hotels, garages, factories, etc.). So I decided I needed to create a layout design of the yard to see how the railway would fit in it. And also to create some designs of buildings. I’ve always had an interest in architecture as well and like to design houses (designed two of the ones I had built and lived in for a few years while bouncing around the country doing jobs for the company I worked for). And my profession was as a software engineer. I’ve loved writing software since high school. So I decided I needed as CAD program to do the design and drafting for the railway and buildings and just to play around with designing real houses. I looked around and decided that the current CAD routines available to me were either too costly, too complex, or too simplistic, or simply didn’t do what I wanted or how I wanted to do things. So I started writing my own basic CAD app using Java via the Netbeans and Eclipse IDEs on a Windows platform, and C# via Visual Studios on the Windows platform. Got pretty far with it in that I could create lines, rectangles, circles, ellipses, free form lines, and arcs. Then created the ability to group entities I had drawn into groups, much like your blocks, and could group the groups into a hierarchy of groups, along with adding new entities to the groups at any level of the group hierarchy. (That’s something you can’t do - yet. It would appear the DXF format does not support that kind of thing. But then, I was writing the results to my own file format in an XML type style (though XML files have a HUGE overhead). But I’d be happy to help you with it. May mean using a different file format or coming up with your own like I did. There is a LOT of overhead and somewhat useless stuff in the DXF format. I wonder if it is even relevant anymore.) But I ran into some minor problems with the graphics libraries in C# that Java solved, so switched over totally to Java. (C# may now have been upgraded to solve the problems, which were primarily in the use of XOR functions to merge colors of new drawables with the background and currently drawn entities.) But the Netbeans IDE is a piece of trash; it has some severe problems and Java is still to me in it’s infancy as a “high-level” language. I’d still much prefer assembly language over anything, but especially Java.

Well, a friend of mine mentioned LibreCAD to me. It’s free and seemed to do a good job for him. So I looked it up. And sure enough, it is pretty good. Could use a few more capabilities (like the grouping I mentioned above, and display of entity properties for info as well as modification purposes needs to be enhanced a lot). But those will come along, I’m sure. Especially if I can help you with them. But then again, it may mean a departure from the accepted standard for graphical file formats. I don’t know what all the types of formats are, so definitely don’t know what they will support. But something out there must support what I want, and hopefully what you will be able to provide in the near future. And if not, is it really that critical to use an antiquated file format? Especially when a lot of things are converting to use XML styles?

Well, thanks for reading this far. And if I can help you out, I will. And not just with what I want, but with things you might want someone else to take a look at doing. Like some “low priority” thing, or some beta testing? Just let me know how I can help.

Thanks, DErik

dxli, 2016-01-29:
@dreappdev 

To build LibreCAD from source, you may follow our wiki:

[Building Howto](http://wiki.librecad.org/index.php/LibreCAD_Installation_from_Source#Alternative:_Building_from_Downloaded_Source_Code)

If you can follow the steps to build, you may continue to play with git and Qt-Creator(IDE for Qt applications), which you can download from qt-project, just install from MacPorts.

Take your time to get used to Qt Creator, which seems to be much easier than Visual Studio to me

dxli, 2016-01-29:
@dreappdev 

another thing, for Java IDE, did you try [IntelliJ](https://www.jetbrains.com/idea/)?

r-a-v-a-s, 2016-01-29:
Rather than creating a new format I suggest seeing if the MakerCAM SVG export can be improved, and implement the ability to import. SVG can also be used with InkScape, which is lacking in CAD style drawing tools. Adobe Illustrator is also lacking (or was before I stopped using it).

Qt Creator tips:
Aside from http://doc.qt.io/qtcreator/creator-editor-locator.html
I heavily use "Switch Header/Source", "Follow symbol under cursor" and "Find usages"
from the text area's context menu.
Also, Ctrl+F has an "Advanced" section that can search all files in the project.
The bar above the text area drops down and has sorted symbols;
and the Outline in the sidebar shows them in order of appearance.
It can also be helpful to use bookmarks. 
http://doc.qt.io/qtcreator/creator-coding-navigating.html

dreappdev, 2016-01-29:
I’ll look at this today.

Thanks, DErik

dreappdev, 2016-01-29:
No, I didn’t. Haven’t heard of it before. I’ll add this to my list of things to check out. But right now I’ve pretty much given up on Java. Swift (for Mac) within the Xcode IDE is what I’m currently looking at to build some other apps (like a financial analysis routine I’ve been working on on various platforms with various languages since 1966 - started with punch cards using FORTRAN on an IBM 1401 at college. That was fun. Fortunately, while a student at college, I was the computer operator for the college’s  math and physics departments, which gave me virtually unlimited access to the system. Unfortunately, I haven’t made much in the stock market yet - mostly still a loser there. But it’s just a hobby.) 

Thanks, DErik

dreappdev, 2016-01-29:
I haven’t looked at the SVG format yet. The DXF format seems formidable enough (nearly 300 pages in the file format description manual!). So I’ll add that to my list of investigation activities.

Also I’ll try downloading Qt Creator and play with it. I recall from Netbeans and VS, they both have the search capability similar to what you describe below. Should be easy to pick it up.

Thanks, DErik

dreappdev, 2016-01-31:
I downloaded MacPorts and installed it. Then attempted to install Qt Creator with:
$ sudo port install gcc49 qt5-creator-mac qt5-mac boost freetype
One thing to note about that is that “xcode-select —install” really does need to be run prior to installing it else gcc49 processing fails because “libgcc" isn’t installed. Also, the name "qt5-creator-mac" has been changed to “qt5-creator” and “qt5-mac” to just “qt5”.

I then tried to load an example app per the Qt Creator web page () to see if it loaded correctly. But the examples were all missing. So I simply went ahead with your instructions in “Get Latest LibreCAD Source Code”. The following appeared to work OK:
    $ sudo port install git-core
    $ mkdir -p ~/github
    $ cd ~/github
    $ git clone https://github.com/LibreCAD/LibreCAD.git
except that “git-core” has been changed to just “git”. I didn’t do the bit regarding updating a previously cloned repository - didn’t have one previously. But the following command to make sure the libraries can be found by “pkg-config”:
   $ echo "QT_CONFIG -= no-pkg-config" >> custom.pro
won’t work because there is no file “custom.pro” to append the line to. At least not in "~/github” or any sub-directories.

I then tried to build LibreCAD with
$ qmake librecad.pro -r -spec macx-g++
but my system could not find "qmake". So I tried doing the following (but changed "gcc48" to "gcc49"):
   $ sudo port install gcc48
   $ sudo port select --set gcc mp-gcc48
but that didn’t do anything regarding "qmake".

So I’m kind of stuck there. Also, the references later in your instructions for building using the script didn’t help - could not find “qmake”. Any ideas on what I did wrong and what I should do next?

Thanks, DErik

dxli, 2016-01-31:
@dreappdev 

after the **git clone** step, need to get into the LibreCAD source folder:
```bash
$ git clone https://github.com/LibreCAD/LibreCAD.git
$ cd LibreCAD/
```
on the qmake question, have a look at `https://github.com/thoughtbot/capybara-webkit/wiki/Installing-Qt-and-compiling-capybara-webkit`

We need to update our wiki, and feel free to help.

dreappdev, 2016-01-31:
OK, Will do. I'll let you know the results.

Thanks, DErik

r-a-v-a-s, 2016-01-31:
Note the wiki is still locked for a move that is theoretically happening.
We can add articles to the GitHub wiki temporarily.

Also it would be more useful to have this talk about building on the forum;
it would keep this issue cleaner and would avoid the information being hidden when this issue is closed.
http://forum.librecad.org/

(hopefuly that sounded like a friendly suggestion instead of a rule nazi dictating)
:v:

dreappdev, 2016-01-31:
The forum is the appropriate place for a discussion of new features or problems. So, no, you don't come off like a nazi dictator. So I'll start using that for such discussions. But so far, it has mainly been you guys helping me get up to speed on setting up a usable environment. Perhaps there is a topic in the forum for that as well. I haven't looked at the forum yet. But though I may look like an ignorant neophyte (truth sometimes hurts), some of my setup experience and activities may be of use to others. Can the emails be moved to the forum (without retyping everything)? Just the essence and procedural updates would be useful, not all the details or extraneous narrative I put in there.

Thanks, DErik

dreappdev, 2016-02-01:
Yes, reading the below referenced page, I see that the binaries for Qt are in /opt/local/libexec/qt5/bin. “qmake” is indeed there, so I added that directory to my PATH variable and attempted to make a new variable QMAKE per the instructions there. This is what I got from that:

Donalds-MBP:LibreCAD DErik$ echo $PATH
/opt/local/bin:/opt/local/sbin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/opt/X11/bin
Donalds-MBP:LibreCAD DErik$ PATH=$PATH:/opt/local/libexec/qt5/bin
Donalds-MBP:LibreCAD DErik$ echo $PATH
/opt/local/bin:/opt/local/sbin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/opt/X11/bin:/opt/local/libexec/qt5/bin
Donalds-MBP:LibreCAD DErik$ which qmake
/opt/local/libexec/qt5/bin/qmake
Donalds-MBP:LibreCAD DErik$ QMAKE=/opt/local/libexec/qt5/bin/qmake gem install capybara-webkit

Fetching: mime-types-data-3.2015.1120.gem (100%)
ERROR:  While executing gem ... (Gem::FilePermissionError)
    You don't have write permissions for the /Library/Ruby/Gems/2.0.0 directory.
Donalds-MBP:LibreCAD DErik$ 

I checked  and sure enough, that directory is not even readable by me (user "DErik"). Only “system” has RW access. How that directory got created in the first place is beyond me. It is owned by root/wheel and so I can’t even modify the access.

Kind of stuck again because I don’t know how to get “admin” or “system” access on a Mac yet. I can on a Windows platform. But Apple seems to think all users are to be severely restricted in what they can do. Maybe not a bad position to take, but really cramps my style.

Thanks, DErik

dxli, 2016-02-01:
@dreappdev 

I have no idea why you would do this:

`$ QMAKE=/opt/local/libexec/qt5/bin/qmake gem install capybara-webkit
`

You should follow LibreCAD instructions instead.

dreappdev, 2016-02-02:
I have started a new topic, 	
"dev environment setup and usage" by Derik,
to continue discussions about the development environment setup and usage. Only discussions directly related to the spline processing should be included in this thread.
Thank you -- DErik

closed 2016-04-02 by dxli