LibreCAD/LibreCAD #639: selecting two polylines and "convert to polyline" leaves stray segments

kfoltman, 2016-02-07:
Steps to reproduce:

1. draw 5 connected lines using Draw / Line / 2 points
2. select first two lines
3. convert to polyline
4. select the remaining 3 lines
5. convert them to polyline
6. select the 2 resulting polylines
7. convert to polyline
8. select the resulting polyline (now it works fine)
9. deselect all (via Ctrl+K) - this part is important, I think
10. select the resulting polyline again (note the different display this time)
11. move it somewhere
12. notice the extra bunch of polyline segments

kfoltman, 2016-02-07:
Actually, step 8 only appears to work fine because some of the objects are still selected after step 7.

So, the shorter steps to reproduce:
1. draw a 2-segment polyline
2. draw a 3-segment polyline starting at the last point of the first one
3. select both
4. convert to polyline
5. mv 0,0 0,100, select "delete original" in the dialog
6. notice the remaining segments of the old polylines

kfoltman, 2016-02-07:
This seems to happen even if converting a single, 3-point polyline to a polyline (supposed to be a null or almost-null operation). It's more obvious to see with antialiasing enabled in settings, because single lines are then drawn in gray and duplicate/overlapping lines are drawn in white (i.e. twice). The other, less convenient way to notice the problem is to use the Info/List entities function to list the vertices of the produced polyline.

I think what may trigger the bug is the fact that in RS_ActionPolylineSegment::convertPolyline, selectedEntity is a polyline (RS2::EntityPolyline), but the entries added to 'remaining' list in the first loop are actually the individual segments of the polyline, i.e. RS2::EntityLine, due to the way iteration works. The end result is that the list of segments to chain contains both entire polyline mixed with polyline segments.

One fix that I've found was to use ResolveNone instead of ResolveAllButInserts in convertPolyline - in that case, polylines are not divided into segments in that iteration, which is probably a good thing, because it's not possible to select single line of a polyline anyway. However, it also means that other entity containers (inserts is one example I can think of) might not work correctly - and I don't even know what the correct behaviour should be in that case.

kfoltman, 2016-02-07:
The same goes for the RS_ActionPolylineSegment::init, it should also be a ResolveNone - otherwise another edge case gets broken (merging two polylines will only add the first segment of one polyline to be merged).

kfoltman, 2016-02-07:
Some edge cases that may be worth testing, and seem to work after changing the iteration mode:

1. single polyline with and without (old method) selection
2. merging two polylines with and without selection
3. merging a polyline and some single lines, with and without selection; partial selection that excludes some of the lines also works
4. adding an arc to an existing polyline
5. converting two connected arcs into a new polyline

r-a-v-a-s, 2016-09-07:
@kfoltman Do you consider this closed?

kfoltman, 2016-09-08:
Yes, it still seems to still work.

closed 2016-09-08 by kfoltman