LibreCAD/LibreCAD #641: Command handling - is there any existing plan to improve it?

kfoltman, 2016-02-07:
I've noticed that handling subcommands within top-level operations (like radius within fillet command) is lacking a mechanism to indicate that the command has been validated, so the UI always reacts with "Unknown command", even though the command has in fact been accepted. Looking at the code, it seems that there's no way to allow command widget to know that the command has in fact been accepted.

Also, the command widget code relies on a tiny hack (detecting a comma and an 'at' sign) to suppress the "Unknown command" message, presumably to allow entering coordinates. I haven't found how the suppression work for single-value arguments yet.

Is anyone planning on doing any changes in that area, or perhaps I should try?

r-a-v-a-s, 2016-02-07:
If you don't see anything regarding an issue here or http://forum.librecad.org/
then there likely is not a plan. I try to be transparent about what I work on by self assigning issues and using the active tag. It's of course smart to ask anyhow. If you are motivated to solve this issue than you are indeed encouraged to try.

r-a-v-a-s, 2016-02-07:
http://sourceforge.net/p/librecad/bugs/?source=navbar is another place to check for issues.

r-a-v-a-s, 2016-02-24:
@kfoltman Your effort is appreciated! :sunglasses:  
If you think you have addressed an issue, then feel free to close it.

closed 2016-02-26 by kfoltman