LibreCAD/LibreCAD #645: No keyboard shortcut for getting into command line

kfoltman, 2016-02-09:
As far as I can remember, Ctrl+M could previously be used to select the command line entry widget. Now it doesn't seem to be working, and that makes it necessary to switch between keyboard and mouse more often.

r-a-v-a-s, 2016-02-09:
Hi, I removed that recently, or didn't bother re-adding it when moving some things around.
I never have understood why it is needed. All typing automatically goes to the command line... does it not? What operating system are you using?

r-a-v-a-s, 2016-02-09:
:bulb:  You need it when you are in the field of another widget...

I'll add it back.

r-a-v-a-s, 2016-02-09:
I implemented it differently; let me know if it works as expected.

kfoltman, 2016-02-09:
Yep, it works.

closed 2016-02-12 by r-a-v-a-s