LibreCAD/LibreCAD #647: Drawing problem, entities randomly missing during zoom

vlachoudis, 2016-02-11:
I've just downloaded the latest master and I while zooming with the mouse I see that randomly some entities are not drawn at all, it depends on the zoom level and the position. Here are some screen shots of the problem
The initial drawing
[image]
After some zooming (the outer white circle disappeared)
[image]
Continue zooming it reappears
[image]

r-a-v-a-s, 2016-02-11:
:white_check_mark: Reproduced

doak, 2018-07-07:
See [this comment](810#issuecomment-403211664) for a new workaround.

sand1024, 2026-09-26:
too old and outdated, not reporoduceable

closed 2026-09-26 by sand1024