LibreCAD/LibreCAD #650: Coredump while trying to offset and delete entity

vlachoudis, 2016-02-12:
I was trying to offset and then undo and delete the created entities and I've got the following coredump. Please also note that offset distance dock was appearing only the first time. On subsequent offset it didn't appear at all.

```
GNU gdb (GDB) Fedora 7.10.1-30.fc23
Copyright (C) 2015 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-redhat-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /home/bnv/git/LibreCAD/unix/librecad...done.
[New LWP 31265]
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib64/libthread_db.so.1".
Core was generated by `/home/bnv/git/LibreCAD/unix/librecad'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  RS_Flags::getFlag (this=this@entry=0x0, f=f@entry=1) at lib/engine/rs_flags.cpp:62
62		return flags&f;
Missing separate debuginfos, use: dnf debuginfo-install OpenEXR-libs-2.2.0-4.fc23.x86_64 appmenu-qt-0.2.6-8.fc23.x86_64 attica-0.4.2-8.fc23.x86_64 bzip2-libs-1.0.6-19.fc23.x86_64 clucene09-core-0.9.21b-15.fc23.x86_64 dbusmenu-qt-0.9.3-0.10.20150604.fc23.x86_64 gamin-0.1.10-18.fc23.x86_64 herqq-1.0.0-10.fc23.x86_64 ilmbase-2.2.0-4.fc23.x86_64 jasper-libs-1.900.1-31.fc23.x86_64 kde-platform-plugin-4.11.22-2.fc23.x86_64 kde-runtime-libs-15.12.1-1.fc23.x86_64 kde-style-breeze-5.5.4-1.fc23.x86_64 kdelibs-4.14.16-1.fc23.x86_64 krb5-libs-1.14-7.fc23.x86_64 libXft-2.3.2-3.fc23.x86_64 libXpm-3.5.11-4.fc23.x86_64 libXtst-1.2.2-5.fc23.x86_64 libgcc-5.3.1-2.fc23.x86_64 libicu-54.1-5.fc23.x86_64 libstdc++-5.3.1-2.fc23.x86_64 libwebp-0.4.4-1.fc23.x86_64 libxml2-2.9.3-2.fc23.x86_64 openssl-libs-1.0.2f-1.fc23.x86_64 qca-2.1.1-2.fc23.x86_64 qtsoap-2.7-11.fc23.x86_64 soprano-2.9.4-8.fc23.x86_64 strigi-libs-0.7.8-11.fc23.x86_64 systemd-libs-222-14.fc23.x86_64
(gdb) bt
#0  0x0000556e93fa6410 in RS_Flags::getFlag(unsigned int) const (this=this@entry=0x0, f=f@entry=1)
    at lib/engine/rs_flags.cpp:62
#1  0x0000556e93f07c8a in RS_Undoable::isUndone() const (this=this@entry=0x0)
    at lib/engine/rs_undoable.cpp:75
#2  0x0000556e93f06a4c in RS_Undo::startUndoCycle() (this=0x556e967c0aa0) at lib/engine/rs_undo.cpp:89
#3  0x0000556e93f7c5b4 in RS_Modification::remove() (this=0x556e967c09a0) at lib/engine/rs_document.h:135
#4  0x0000556e93f7c5b4 in RS_Modification::remove() (this=this@entry=0x7ffd2d3da5e0)
    at lib/modification/rs_modification.cpp:99
#5  0x0000556e94005e93 in RS_ActionModifyDelete::trigger() (this=0x556e97a26e90)
    at actions/rs_actionmodifydelete.cpp:58
#6  0x0000556e93f546a0 in RS_EventHandler::setCurrentAction(RS_ActionInterface*) (this=0x556e9640ead0, action=action@entry=0x556e97a26e90) at lib/gui/rs_eventhandler.cpp:477
#7  0x0000556e93f55da3 in RS_GraphicView::setCurrentAction(RS_ActionInterface*) (this=
    0x556e967c8c00, action=action@entry=0x556e97a26e90) at lib/gui/rs_graphicview.cpp:254
#8  0x0000556e94040c7c in QG_ActionHandler::setCurrentAction(RS2::ActionType) (this=0x556e96573560, id=<optimized out>) at ui/qg_actionhandler.cpp:905
#9  0x00007f805f874a80 in QMetaObject::activate(QObject*, QMetaObject const*, int, void**) (sender=sender@entry=0x556e965419e0, m=m@entry=0x7f80608c07a0 <QAction::staticMetaObject>, local_signal_index=local_signal_index@entry=1, argv=argv@entry=0x7ffd2d3da800) at kernel/qobject.cpp:3567
#10 0x00007f805fdc89d2 in QAction::triggered(bool) (this=this@entry=0x556e965419e0, _t1=false)
    at .moc/release-shared/moc_qaction.cpp:276
#11 0x00007f805fdc9d33 in QAction::activate(QAction::ActionEvent) (this=0x556e965419e0, event=event@entry=QAction::Trigger) at kernel/qaction.cpp:1257
#12 0x00007f805fdca28c in QAction::event(QEvent*) (this=<optimized out>, e=<optimized out>)
    at kernel/qaction.cpp:1183
#13 0x00007f805fdce93c in QApplicationPrivate::notify_helper(QObject*, QEvent*) (this=this@entry=0x556e96336790, receiver=receiver@entry=0x556e965419e0, e=e@entry=0x7ffd2d3daaf0) at kernel/qapplication.cpp:4565
#14 0x00007f805fdd5796 in QApplication::notify(QObject*, QEvent*) (this=0x7ffd2d3db9d0, receiver=0x556e965419e0, e=0x7ffd2d3daaf0) at kernel/qapplication.cpp:4351
#15 0x00007f805f8608dd in QCoreApplication::notifyInternal(QObject*, QEvent*) (this=0x7ffd2d3db9d0, receiver=0x556e965419e0, event=event@entry=0x7ffd2d3daaf0) at kernel/qcoreapplication.cpp:955
#16 0x00007f805fe05738 in QShortcutMap::dispatchEvent(QKeyEvent*) (event=0x7ffd2d3daaf0, receiver=<optimized out>) at ../../src/corelib/kernel/qcoreapplication.h:231
#17 0x00007f805fe05738 in QShortcutMap::dispatchEvent(QKeyEvent*) (this=this@entry=0x556e963368b8, e=e@entry=0x7ffd2d3dae10) at kernel/qshortcutmap.cpp:887
#18 0x00007f805fe05927 in QShortcutMap::tryShortcutEvent(QObject*, QKeyEvent*) (this=0x556e963368b8, o=o@entry=0x7ffd2d3dba00, e=e@entry=0x7ffd2d3dae10) at kernel/qshortcutmap.cpp:367
#19 0x00007f805fdd6b6e in QApplication::notify(QObject*, QEvent*) (this=0x7ffd2d3db9d0, receiver=0x7ffd2d3dba00, e=0x7ffd2d3dae10) at kernel/qapplication.cpp:3989
#20 0x00007f805f8608dd in QCoreApplication::notifyInternal(QObject*, QEvent*) (this=0x7ffd2d3db9d0, receiver=receiver@entry=0x7ffd2d3dba00, event=event@entry=0x7ffd2d3dae10) at kernel/qcoreapplication.cpp:955
#21 0x00007f805fdcd2e6 in qt_sendSpontaneousEvent(QObject*, QEvent*) (event=event@entry=0x7ffd2d3dae10, receiver=receiver@entry=0x7ffd2d3dba00) at ../../src/corelib/kernel/qcoreapplication.h:234
#22 0x00007f805fdcd2e6 in qt_sendSpontaneousEvent(QObject*, QEvent*) (receiver=receiver@entry=0x7ffd2d3dba00, event=event@entry=0x7ffd2d3dae10) at kernel/qapplication.cpp:5563
#23 0x00007f805fe74463 in QKeyMapper::sendKeyEvent(QWidget*, bool, QEvent::Type, int, QFlags<Qt::KeyboardModifier>, QString const&, bool, int, unsigned int, unsigned int, unsigned int, bool*) (keyWidget=keyWidget@entry=0x7ffd2d3dba00, grab=grab@entry=110, type=QEvent::KeyPress, code=16777223, modifiers=..., text=..., autorepeat=false, count=1, nativeScanCode=119, nativeVirtualKey=65535, nativeModifiers=0)
    at kernel/qkeymapper_x11.cpp:1866
#24 0x00007f805fe74972 in QKeyMapperPrivate::translateKeyEvent(QWidget*, _XEvent const*, bool) (this=0x7ffd2d3db380, keyWidget=0x7ffd2d3dba00, event=0x556e00000000, grab=<optimized out>)
    at kernel/qkeymapper_x11.cpp:1836
#25 0x00007f805fe4e5a6 in QApplication::x11ProcessEvent(_XEvent*) (this=0x7ffd2d3db9d0, event=event@entry=0x7ffd2d3db380) at kernel/qapplication_x11.cpp:3556
#26 0x00007f805fe77249 in x11EventSourceDispatch(GSource*, GSourceFunc, gpointer) (s=0x556e96337fb0, callback=0x0, user_data=0x0) at kernel/qguieventdispatcher_glib.cpp:148
#27 0x00007f805d5f5e3a in g_main_context_dispatch (context=0x556e96336cf0) at gmain.c:3154
#28 0x00007f805d5f5e3a in g_main_context_dispatch (context=context@entry=0x556e96336cf0) at gmain.c:3769
#29 0x00007f805d5f61d0 in g_main_context_iterate (context=context@entry=0x556e96336cf0, block=block@entry=1, dispatch=dispatch@entry=1, self=<optimized out>) at gmain.c:3840
#30 0x00007f805d5f627c in g_main_context_iteration (context=0x556e96336cf0, may_block=may_block@entry=1)
    at gmain.c:3901
#31 0x00007f805f890e8e in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) (this=0x556e96336be0, flags=...) at kernel/qeventdispatcher_glib.cpp:450
#32 0x00007f805fe773e6 in QGuiEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) (this=<optimized out>, flags=...) at kernel/qguieventdispatcher_glib.cpp:207
#33 0x00007f805f85f151 in QEventLoop::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) (this=this@entry=0x7ffd2d3db760, flags=...) at kernel/qeventloop.cpp:149
#34 0x00007f805f85f4c5 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) (this=this@entry=0x7ffd2d3db760, flags=...) at kernel/qeventloop.cpp:204
#35 0x00007f805f865069 in QCoreApplication::exec() () at kernel/qcoreapplication.cpp:1227
#36 0x0000556e93e50ecc in main(int, char**) (argc=1, argv=<optimized out>) at main/main.cpp:365
(gdb) q
```

dxli, 2016-02-16:
@vlachoudis 

just want to confirm: this bug is from the latest master HEAD.

vlachoudis, 2016-02-16:
Yes, at least the HEAD master 4 days ago

r-a-v-a-s, 2016-02-24:
Does this appear to be fixed?

vlachoudis, 2016-02-25:
I cannot reproduce it anymore.
BTW offset in a rectangle leaves it open on the starting point

r-a-v-a-s, 2016-03-06:
I confirm that; although Create Equidistant Polylines seems to work as expected.
It's probably worth opening another issue.

closed 2016-03-06 by r-a-v-a-s