LibreCAD/LibreCAD #652: Snapping on hatch lines

vlachoudis, 2016-02-12:
Hatch lines are included when snapping is selected, creating a confusion.
In my opinion should be ignored from the snapping.

dxli, 2016-02-16:
@vlachoudis :+1: 

I remember long time ago. There was a feature request to enable snapping on blocks, and enabling snapping on blocks causes snapping of hatch lines.

vlachoudis, 2016-02-17:
Snapping on blocks is a good thing. Snapping to hatch lines leads to confusion :)

dxli, 2016-02-19:
@r-a-v-a-s 

I like the idea of relying more on polymorphism. Actually, using of our own rtti() seems to be awkward from the beginning.

Another thought, we may want to port some design from V3.

r-a-v-a-s, 2016-02-22:
It seems like having a hatch container skips the need to check during intersection analysis.
It also seems like could make the toggling of the draft mode cleaner.
However, I don't have these systems in my mind.

dxli, 2016-02-22:
first, let's revert the commit 367d18e2610

r-a-v-a-s, 2016-02-24:
Introducing the pseudobug label...
for bugs that could be features. :sunglasses:

r-a-v-a-s, 2016-02-24:
The latest commit seems to produce the same issue with previews not showing, and using hatch with a rectangle:
`RS_Hatch::update: invalid contour in hatch found`

It looks like `update` calls `validate` which calls `optimizeContours` which calls `getNearestEndpoint`.

I found that reverting to `IgnoredModifications` for `getNearestEndpoint` solves the two issues;
and adding `RS2::EntityHatch` to `IgnoredModifications` works to avoid snapping.

The same trick doesn't seem to work for `getNearestIntersection`; 
and `ignoredSnap` avoids almost all intersections 
except for those with the rectangle the hatch is applied to.

It's too late for me to understand that function... my mind cringes at:
`for (RS_Entity* en = firstEntity(RS2::ResolveAllButTextImage);`

We will see what tomorrow brings... :v:

shawncurry, 2019-05-19:
> Introducing the pseudobug label...
> for bugs that could be features. 😎

Deactivating it, because I'm collapsing the categories to make them more useful.  😎

fa201, 2020-04-13:
It is not completely fixed.
I cannot snap on hatch lines entities, extremities and center but I can snap on hatch line intersection with the part.