LibreCAD/LibreCAD #657: Conflicting menu hotkeys

kfoltman, 2016-02-13:
_Draw, _Dimension and _Dockwidgets

_Select and _Snap

To be fair, several of those top-level menus could work equally well as submenus. For example, Toolbars and Dockwidgets tend to be placed in the View menu, next to related items such as Statusbar.

r-a-v-a-s, 2016-02-13:
Several of them were submenus. ;-]
I expanded the menubar to promote user awareness;
several people have needed to be told where to look to toggle toolbars.
However, everything you said is valid.

What operating system are you on?
Can you give me some examples of how you use these menu hotkeys?
Perhaps a more efficient system can be conceived.

kfoltman, 2016-02-13:
I'm using Linux (Debian with XFCE environment, to be precise). I'm using the menu bar to access functions that I don't use frequently enough to remember the commands. Usually it's faster to find, say, Modify/Trim by Alt+M / T / Enter (which has another conflict, by the way) than by using a mouse - at least for a person who uses the keyboard as the primary input device.

Besides, having conflicting shortcuts is typically considered a minor UI bug - not sure about other OSes, but in Linux it causes the menu behaviour to be non-deterministic, for example pressing Alt+D the first time selects the first menu, but if you close the menu then try it again, it selects the second menu instead etc.. The typical "proper" way is to use a different non-conflicting letter for the shortcut (S_nap) - it looks worse, but at least works predictably. But it can't be done easily for all options, because some menu names should never be changed (File, Edit, View, Window, Help) and some don't have enough unique letters to be able to pick something non-unique.

It's considered important enough that some tools (like MSVC) used to have an option to detect shortcut conflicts - not sure if they still have, as I haven't used those for years.

Maybe it would help to print the command name in the command line window when it's selected from the menu, to make it easier to learn the commands and stop using the menus altogether.

Not sure what can be done to make it easier for the users to find the features - maybe having them in similar places to popular programs (Firefox etc.) could work, but then again I know enough people to suspect that it won't be the case.

r-a-v-a-s, 2016-02-13:
> "Maybe it would help to print the command name in the command line window when it's selected from the menu, to make it easier to learn the commands and stop using the menus altogether."

I recently added that feature, but only to the modify actions... more work to do.
https://github.com/LibreCAD/LibreCAD/issues/570

I have plans to make it so you can define your own hotkey for any action.
https://github.com/LibreCAD/LibreCAD/issues/623

I used to be pretty passionate about hotkeys;
however, I find the combination of the custom toolbar 
and the recent actions context menu to be efficient for my needs.  
http://forum.librecad.org/GUI-update-td5712246.html

I'm working on improving the custom toolbar;
it's currently too inconspicuous.
https://github.com/LibreCAD/LibreCAD/issues/631

1. make a blank custom.txt file in the folder you prefer 
2. in application preferences (paths tab) choose the txt file you made 
3. restart librecad 
4. look for the ± icon

You can add or remove the last triggered action by pressing F2 
or pressing the always ± icon. 

The menubar is also a work in progress;
if there is anything you desperately want changed, then feel free to point it out.
If you want a hotkey for a specific action, then feel free to suggest one.

r-a-v-a-s, 2016-03-05:
@kfoltman I'm on OS X right now (I need a new table to set up my PC), which doesn't have the concept of accelerator keys. If you are interested in working on this issue, then look in lc_actionfactory.
You simply need to place an `&` in front of the letter you want to use.
example: `action = new QAction(tr("&Trim"), tool_group);`

kfoltman, 2016-03-08:
@r-a-v-a-s Yes, I've started working on it, but the drawback is that this kind of changes invalidate the translations, so I'll have to do some scripting work to cope with that. Also, the Modify menu contains enough items to make it impossible to have unique accelerators. Still, I should be able to get the most common accelerators sorted out.

r-a-v-a-s, 2016-03-09:
I wonder if there is a way to remove the `&` from the whole translation process.