LibreCAD/LibreCAD #658: zooming on osx

r-a-v-a-s, 2016-02-15:
On OS X the mouse wheel events trigger scrollbar movement,
and zooming requires that the command button is held.
I say that zooming is the priority; and it's inconsistent with the behavior on other operating systems.

The second issue is apple's crazy concept of "natural scrolling",
which is the inverse of scrolling on Windows... I don't know about other operating systems.
We are currently compensating for this, which I think is a mistake.
I say we let it be determined by the state of the operating systems "natural scrolling" option,
and provide our own option to invert.

Technically the associate code is tied to the use of `QWheelEvent::pixelDelta()`;
however, we also use `QWheelEvent::phase()` which is only supported on OS X.

r-a-v-a-s, 2016-02-16:
After more review I realized that the relevant code was intended for high-resolution trackpads,
but because it was missing `e->source() == Qt::MouseEventSynthesizedBySystem`
normal wheel events were being processed as trackpad events on OS X.
Both issues are solved with this fix;
however, I still like the idea of an inversion option,
and also a zoom factor / speed option.

r-a-v-a-s, 2016-02-16:
And I'm still thinking about http://forum.librecad.org/ctrl-mouse-wheel-td5712660.html

rvt, 2016-05-14:
Just to give my 2 cents, in LibreCAD 3 scrollwheel will zoom in/out. For my feeling it allows you to navigate very fast over a drawing zooming in/out at different locations.
That said, I guess (at least in LibreCAD 3) that should be a setting, and not hard coded so people can change it themselve

r-a-v-a-s, 2016-05-14:
Users can select trackpad in Device Options if they want the vertical scrolling.
I don't think it's important enough to make a dedicated option,
because it was never intended behavior for mouse.

If anyone else wants to do the work, I'm not opposed to it.

r-a-v-a-s, 2016-05-23:
Regarding the 2.0 branch:
on OS X with Qt 5.6 
`e->source() == Qt::MouseEventSynthesizedBySystem` is true when using a Mac mouse...
therefore, the mouse wheel manipulates the vertical scrollbar.
I originally tested with a PC mouse so I didn't realize this...
PC mice are considered real mice on Mac...  :astonished: 
This seems like a Qt bug, since a Mac mouse would in theory send mouse events...
I don't really want to do the work to port the 2.1.0 solution.
I'm tempted to just remove the whole "high resolution trackpad" section.
Does anyone have another solution?

r-a-v-a-s, 2016-05-23:
Forget it... I'll port my solution from 2.1.0...

closed 2016-05-23 by r-a-v-a-s