LibreCAD/LibreCAD #660: undo & and trim2 crash

r-a-v-a-s, 2016-02-16:
1. create two non-intersecting line segments, but the lines would intersect
2. use trim2 on them
3. undo
4. repeat step 2 :triangular_flag_on_post: crash

[image]
 
[image]

dxli, 2016-02-17:
@r-a-v-a-s 

doesn't creash for me with 75be87a5f9

r-a-v-a-s, 2016-02-17:
It still crashes.

r-a-v-a-s, 2016-02-17:
The line segments can also be intersecting; it doesn't matter.
You can also use trim or trim2.

r-a-v-a-s, 2016-02-17:
`for(auto u: l->undoables)` 
and it says `u` is 0x0

 Why is undoList a vector of shared_ptr? What does the shared_ptr accomplish?

How is it that `l->undoables` is accessed in this loop when it is a private member?

dxli, 2016-02-17:
need to figure out why u could be nullptr. The undoCycle implementation doesn't accept any nullptr in undo cycles.

because we keep going back and forth by the undo pointer (pointing to undo cycles), and using raw pointers could lead to memory leak.

I'm wondering whether objects should be used instead of smart pointers.

dxli, 2016-02-17:
it still doesn't crash here.

Can you share a dxf to show the crash?

the u is from undoCycle, but the undocycle implementation doesn't accept a nullptr into cycle. very puzzling

We still have to fix any crashing bugs at high priorities

r-a-v-a-s, 2016-02-17:
I don't think trim has anything to do with it actually.

1. create a circle
2. use scale or mirror on it (keep original)
3. undo
4. repeat step 2 :triangular_flag_on_post: crash

dxli, 2016-02-17:
still couldn't crash it.

* start LC;
* command line: circle
* 0,0
* 10
* scale
* sa
* 1,1
* Accept scale dialog with "Keep original"
* undo
* repeat from scale

I tried quite many times here, and everything just works as expected

r-a-v-a-s, 2016-02-17:
It crashes with every attempt for me.
It's the same error every time, and you can see my screenshots.
What information are you hoping to acquire?
I think answering my questions is the place to start.

dxli, 2016-02-17:
@r-a-v-a-s 

we are not running the same code. Have a look at d518aa9b4f2056

the first-aid fix doesn't explain why u is nullptr though.

Do you mean the shared_ptr part? what I wanted was an opaque pointer to reduce header dependency. Then:

* raw pointer, potential memory leak
* unique_ptr, would disable any copying, and have to implement move actions
* shared_ptr, lazy implementation

maybe, it's right time implement unique_ptr (for efficiency over shared_ptr).

Need to debug the nullptr occurrence first.

r-a-v-a-s, 2016-02-17:
My other question was about the fact that `for(auto u: l->undoables)` is accessing a private member.
I was using the latest commit from the master branch; and the commit I just made solved the issue.
I'm not sure how it was working for you... or at all...

closed 2016-02-24 by r-a-v-a-s