LibreCAD/LibreCAD #661: Invalid hatch area

Flajr, 2016-02-22:
I created rectangle to avoid non closed contour (and other tools) to get hatching work but version _2.0.7+20160219.344-0~ubuntu14.04.1_ on Lubuntu 14.04 x64
gives this every time.
`Invalid hatch area. Please check that the entities chosen form one or more closed contours.`

Version 2.0.2 seems work smoothly, so question is why hatching can't get to work on 2.0.7?

r-a-v-a-s, 2016-02-22:
Technically what you are using is far beyond 2.0.7,
but the master branch (most recent code)
is still being labeled as 2.0.7 for some reason.
In short you are using a 2.1.0 beta version 
that recently introduced this bug when trying to fix another.
https://github.com/LibreCAD/LibreCAD/issues/652

We can revert the changes, since the introduced bugs are worse than the fixed one.

@dxli Have you thought of any alternatives?

Flajr, 2016-02-22:
Thanks for reply!
So awhile I will stick with official 14.04 repo. Now it's my best option.
Many Thanks.

dxli, 2016-02-22:
@r-a-v-a-s 

do you think 2.1 is stable enough for a tag of 2.1.0-alpha1 ?

the current SCM-version from git is by the latest ancestor tag (which is 2.0.7) for the 2.1 branch.

r-a-v-a-s, 2016-02-22:
@dxli After we deal with this issue I think that would be appropriate.

dxli, 2016-02-23:
@r-a-v-a-s 

thanks! I will try again more carefully.

A quick solution would decouple snapping tests and modification tests.

Hatch allows modification (to generate hatch), but without snapping.

closed 2016-02-23 by Flajr