LibreCAD/LibreCAD #664: 'cut' mode SplinePoints (the ones with control points only) don't snap on start/end

kfoltman, 2016-02-27:
1. Draw a spline using the 'through the points' option
2. Save to DXF
3. Close and load from DXF
4. Try to draw a 2-point line from start point of the SplinePoints to the end point of the SplinePoints using endpoint snap -> it doesn't snap to the endpoints

The bug seems to be in LC_SplinePoints::getStartPoint()/ LC_SplinePoints::getEndPoint() - they always check splinePoints instead of using getPoints() to pick the relevant array.

r-a-v-a-s, 2016-03-02:
dxli is busy this week and you already understand the issue more than I do.
Please join the team and help us get ready for 2.1.0 :sunglasses:

kfoltman, 2016-03-02:
Thanks - I've created a pull request just for extra safety :)

r-a-v-a-s, 2016-03-02:
:+1: Feel free to label and self assign this issue before you close it.

closed 2016-03-02 by kfoltman