LibreCAD/LibreCAD #666: Hatch does not work.

pasha-pivo, 2016-03-02:
At all. Even if you make a square with the tool "Rectangle", the program will not be able to hatch it.
The next message will be shown in any way:

> Invalid hatch area. Please check that the entities chosen form one or more closed contours.

The entities are chosen and the contours are definitely closed. I tried to check it by applying "trim two" to all corners of the figure, but without any result.
Is there anything that can help to find the solution? 

OS Ubuntu 14.04, build 2.0.7+20160224.346-0~ubuntu14.04.1

r-a-v-a-s, 2016-03-02:
That was fixed several days ago. The ubuntu ppa needs to be updated.

r-a-v-a-s, 2016-03-02:
We don't have any active developers using ubuntu that I know.
I emailed Scott Howard yesterday... no response yet.
I don't know who uploaded the last version; it just says "librecad-dev".
https://launchpad.net/~librecad-dev

Help wanted...

pasha-pivo, 2016-03-02:
Got it. Building from source right now and waiting for the updates.
Thank you very much for the response!

r-a-v-a-s, 2016-03-12:
@pasha-pivo Did you build from source using the instructions here https://github.com/LibreCAD/LibreCAD/wiki/Build-from-source#users-of-ubuntudebian-and-derivatives
?

I want to make sure that is up to date.

For building with qt5 I found:
- http://packages.ubuntu.com/xenial/qtbase5-dev
- http://packages.ubuntu.com/xenial/qt5-default
- http://packages.ubuntu.com/xenial/qttools5-dev

I'm not sure which to recommend; If you can build with qt5 let us know.

pasha-pivo, 2016-03-12:
I used instructions here[http://wiki.librecad.org/index.php/LibreCAD_Installation_from_Source](url) but there were some missed dependencies, I found the solution somewhere on Internet, I don't remember where. And I don't remember what version of  qt was used. Honestly, I am not a very skilled builder. 
Anyway, after it was done, it worked well.

closed 2016-03-15 by r-a-v-a-s