LibreCAD/LibreCAD #668: SegFault when opening a file with librecad in Linux Shell and my solution

Xia-Weiwen, 2016-03-03:
Hi, all!
librecad crashed when I was trying to open a dxf file in Linux Shell, and segmentation fault was reported.
What I did is like:
`$ ./librecad test.dxf`
I found that the crash happens when executing line 1512 in librecad/src/main/qc_applicationwindow.cpp
`geo=activedMdiSubWindow->geometry();`
I guess it's because activedMdiSubWindow is actually a null pointer when loading the file at startup, but the if() before this line does not work without initialization of the pointer. So I initialize this pointer at line 292 in qc_applicationwindow.h like this
`QMdiSubWindow* activedMdiSubWindow=nullptr`
Now it seems to be OK.

r-a-v-a-s, 2016-03-03:
It was fixed by https://github.com/LibreCAD/LibreCAD/commit/e1b42aa0c555b35268ef604b919d1f28a062ab6c
in the master branch.

I don't think we added it to the 2.0 branch; thanks for the reminder. :sunglasses:

Xia-Weiwen, 2016-03-03:
that's great :)

closed 2016-03-03 by Xia-Weiwen