LibreCAD/LibreCAD #670: Updated Linux (Ubuntu 15.10) packages (2.0.9)?

poVoq, 2016-03-09:
Seems like there is currently only a good support for Windows, which the Linux repositories really lagging behind.
I managed to get the Ubuntu 14.04 2.0.8 packages to work on my 15.10 system, but only with workarounds and it seems to crash a lot (probably because 15.10 doesn't have the right version of "libmuparser2").

Would be great if someone would update the ppa with some 2.0.9 packages for 15.10 and 16.04. Thanks!

poVoq, 2016-03-09:
Ok, seems like this was answered here:
https://github.com/LibreCAD/LibreCAD/issues/666

I'll leave this open until a maintainer for the linux builds is hopefully found. Thanks!

r-a-v-a-s, 2016-03-09:
I used the "Contact this team's admins" feature yesterday. https://launchpad.net/~librecad-dev
No response yet...
I don't like that none of the activate developers have access to this ppa.
It might be time to start a new one.

Here are some links that might help someone get started:

https://github.com/LibreCAD/LibreCAD/wiki/Build-from-source

- https://help.launchpad.net/Packaging/PPA
- https://help.launchpad.net/Packaging/PPA/BuildingASourcePackage
- https://wiki.ubuntu.com/PbuilderHowto
- https://www.debian.org/doc/manuals/debian-faq/ch-pkg_basics.en.html

r-a-v-a-s, 2016-03-11:
@maqifrnswa I just noticed you are part of the developers team, but are not subscribed to the repository. 
We need a 2.0.9 release from the 2.0 branch, and a new release from the master branch.
2.0.7+20160224.346-0 has some nasty bugs...

Symbian9, 2016-03-13:
Dont forget to make 2.0.9 build for Ubuntu 14.04 too!

r-a-v-a-s, 2016-03-13:
I recommend learning how to build from source.

We need Ubuntu users who can build from source using Qt5 to edit our wiki articles.

https://github.com/LibreCAD/LibreCAD/wiki/Build-from-source

http://wiki.librecad.org/index.php/Build_from_source

I think you would need to use one of these:

- http://packages.ubuntu.com/xenial/qtbase5-dev
- http://packages.ubuntu.com/xenial/qt5-default
- http://packages.ubuntu.com/xenial/qttools5-dev

maqifrnswa, 2016-03-15:
Hi there - looking at this now! will get you a 2.0.9 release today

r-a-v-a-s, 2016-03-15:
@maqifrnswa Thanks! :sunglasses:

maqifrnswa, 2016-03-15:
Updated the wikis too.
1) Qt5 packages needed are: qtbase5-dev libqt5svg5-dev qttools5-dev
qtchooser qttools5-dev-tools

2) You also should install the qt5-default package or run "qtchooser
-qt5" before running qmake if you have both qt4 and qt5 on your
system.

3) everything else should juts work! Package is built on Debian.

r-a-v-a-s, 2016-03-15:
Thanks a lot. :bow: 
You're a hero.
:sunglasses:

poVoq, 2016-03-16:
So no new Ubuntu packages?

Anyways, thanks a lot for looking into it.

r-a-v-a-s, 2016-03-16:
It shows packages from today on both https://launchpad.net/~librecad-dev

But they are tagged 2.0.8,
instead of 2.0.9 and 2.1.0-alpha.

maqifrnswa, 2016-03-16:
Those are both 2.0.8. It takes about a day after uploading to Debian
for it to trickle down everywhere. Within a few hours they will be
2.0.9. In fact, I just checked, the builds started already - so maybe
within 30 mins you'll see new versions.

2.1.0~alpha will show up as 2.0.9+20160316. Everything is automated
and the tags are chosen to be the most recent release + day of the
last code commit.

maqifrnswa, 2016-03-16:
Also requested that the next version of Ubuntu include at least 2.0.9
https://bugs.launchpad.net/bugs/1558067

poVoq, 2016-03-17:
Ok works now, but that it shows up as 2.0.8 +20160315 or something like that in the repository, but is in fact a 2.1.0 build is a bit confusing.

closed 2016-03-17 by poVoq