LibreCAD/LibreCAD #671: Print preview shows a big black circle

jirislaby, 2016-03-09:
With this minimalistic example:
[http://www.fi.muni.cz/~xslaby/sklad/black.dxf]

When I click on the print preview, the line in the middle is rendered as a big black circle covering everything around.

I have librecad-2.0.9-34.2.x86_64 from openSUSE Tumbleweed.

Rupicapra-rupicapra, 2016-03-10:
Hello,
Does setting a pre-defined paper size (like a4, letter…) in the « Current Drawing Preferences » and line widths solves the problem ? (Setting these parameters on my computer solves it)

Rupi

jirislaby, 2016-03-10:
Yes, setting paper to A4 solves the problem. But where did the bad setting come from? (I copied the line from a DWG import).

jirislaby, 2016-03-10:
And what is so special about the one line? The others are rendered correctly...

Rupicapra-rupicapra, 2016-03-10:
I don't really know where that setting comes from… Maybe a dev who was involved into DWG import could answer.

The line is in a layer with a width set to « Default », while the other are in the layer 0 (width = 0.0 mm). I don't know exactly what that « Default » represents, but it’s a value that was too big for the initial paper size (width & height < 1/1000 mm).

sand1024, 2026-09-26:
can't reproduce, too old issue

closed 2026-09-26 by sand1024