LibreCAD/LibreCAD #673: lengthen does not wrod the second time

jirislaby, 2016-03-14:
I have to always modify the length for the lenghten tool. Otherwise, nothing happens to lines.

I sometimes set length to 1000 and use it. Then I use another tool. Then I want to use lengthen again, with the same length, but I have to modify length somehow.

jirislaby, 2016-03-14:
I have librecad-2.0.9-34.2.x86_64 from opensuse,

dxli, 2016-03-18:
My test show lengthen works properly.

Please supply a detailed procedure to show this bug.

jirislaby, 2016-03-22:
I repeat:
I sometimes set length to 1000 and use it. Then I use another tool. Then I want to use lengthen again, with the same length, but I have to modify length somehow.

Draw two lines. Click lengthen, enter 100. Click one line. It is lengthen. Press escape. Click lengthen. Click on the second line. Nothing happens.

dxli, 2016-03-23:
@jirislaby 

please do provide detailed/exact procedure to reproduce the bug:

Example:

* start librecad;
* command line: line
* when prompted "Specify first point", type in coordinates: *0, 0* , and press enter;
* at prompt "Specify next point", type in coordiante: * @10, 10*;
* at prompt "Specify next point", type in coordiante: * @-10, 10*;
* in command line, type in command *lengthen*;
* in the option widget within the top widget area, type in length *100* and press enter;
* click on the line from 0,0 to 10, 10, close to the 10, 10 end point.

Expected results: the line (0,0)-(10,10) is extended by 50 from the (10, 10) end.

dxli, 2016-04-02:
no feedback, closing this one.

closed 2016-04-02 by dxli