LibreCAD/LibreCAD #681: "Create Equidistant Polyline" does not preserve "color by layer" attribute

jborme, 2016-03-18:
When creating an  equidistant polyline, the newly created polyline has visually the same color as the original. However, it seems the visual color was forced set to that particular value. If the original layer had "color by layer" attribute, it was not preserved.

Instructions:
1. Create a polyline, make sure it has the "color by layer" attribute.
2. Create an equidistant polyline.
3. Change the layer of the newly created polyline. 

Result: the polyline will keep its original color, not following the color of its new layer.

dxli, 2016-04-02:
obsoleted by #682

closed 2016-04-02 by dxli

jborme, 2018-05-06:
Here is a proposed correction to the bug:

rs_actionpolylineequidistant.cpp:
-newPolyline->setPen(((RS_Polyline*)originalEntity)->getPen());
+newPolyline->setPen(((RS_Polyline*)originalEntity)->getPen(false));

If getPen() is set to resolve = false, then the correct argument "by layer" is copied. By default, getPen assumes true, which then resolves to a particular color, losing the "by layer" / "by block" attributes that the user set in the original polyline.

I can make this a pull request if it helps.