LibreCAD/LibreCAD #685: Opening a DXF leaks memory

strk, 2016-03-19:
I tried opening a 64MB DXF file which succeeded with Librecad process using about 1.5GB of RAM.
On _closing_ the document those 1.5GB of RAM remained in use, so I guess there's a memory leak somewhere. Unfortunately I cannot share the DXF.

r-a-v-a-s, 2016-03-19:
We have two memory leaks. http://librecad.github.io/static-analyzer-reports/
One is in libdxfrw....
Help is welcome; I'd like to finish a project instead of getting side tracked by yet another bug.

r-a-v-a-s, 2016-03-19:
If the loop condition is really always false, then it seems like there is another bug.
Having the return inside the while loop also complicates the issue.

r-a-v-a-s, 2016-03-19:
Regarding the other leak, I don't understand the purpose of the bracket syntax `delete[]SectionsMapData;`

r-a-v-a-s, 2016-03-19:
Answering my last comment
http://stackoverflow.com/questions/4670782/the-difference-between-delete-and-delete-in-c

r-a-v-a-s, 2016-03-19:
I eliminated that one. https://github.com/LibreCAD/LibreCAD/commit/a11d9257124d4a63cabfeef339bf41153db19ea3

Another complication of the other is `pl->appendVertex(v);`
which adds it to drw_polyline.vertlist

the destructor drw_polyline:

```
    ~DRW_Polyline() {
        while (!vertlist.empty()) {
           vertlist.pop_back();
         }
    }
```

The logic really needs to be cleaned, rather than just adding a delete statement.

strk, 2016-03-20:
On Sat, Mar 19, 2016 at 11:23:17AM -0700, ravas wrote:

> the destructor drw_polyline:
> 
> ```
>     ~DRW_Polyline() {
>         while (!vertlist.empty()) {
>            vertlist.pop_back();
>          }
>     }
> ```
> 
> The logic really needs to be cleaned, rather than just adding a delete statement.

Until a refactoring is done, adding the delete statement wouldn't be bad.
Rather, why is that convoluted loop used rather than a straight scan ?
Are there other threads expected to possibly using that container during
the destructor of that object ?

dxli, 2016-03-22:
@strk I tried to use shared_ptr here.

lazy implementation

Can you build and test to see whether it helps: https://github.com/dxli/LibreCAD/tree/memory

r-a-v-a-s, 2016-03-22:
@dxli He has not given us feedback regarding the latest code yet. https://github.com/LibreCAD/LibreCAD/commit/f7b4421b9d608d5bcb7b45ee5f8e6add81f7e73d

Why don't you think that was the fix?

strk, 2016-03-22:
FYI: the "official tester" for these leaks is a friend of mine who cannot build librecad himself, but he is prompted for auto-upgrades from the daily build PPA. Were any of those commits included in the daily build ?

dxli, 2016-03-22:
Hi ravas,

it's a little bit risky to push a big patch set without a review.

The current situation:

1, Rallaz wants to keep dxfrw without C++11;

2, the raw pointer usage in dxfrw is causing massive memory leak. Anyway, I
guess after we fix dxfrw, we will see other memory leaks from other places;

3, using shared_ptr seems to be an off-the-shelf fix here, but do we have
any situation of memory leak by shared_ptrs? like circular references.

Do you have the time to review what I did with my "memory" branch?

Best,

Dongxu

strk, 2016-03-22:
On Tue, Mar 22, 2016 at 06:18:54AM -0700, Dongxu Li wrote:

> 1, Rallaz wants to keep dxfrw without C++11;

I've just seen C++11 pushed in the master branch,
was this comment obsoleted by later decision ?

dxli, 2016-03-22:
Hi Sandro,

Yes, C++11 was already in dxfrw before we started working on this memory
leak issue.

I was probably the one to blame for bringing in C++11, and that's why I was
aware of of rallaz's intention to keep dxfrw C++11 free.

However, as C++14 is now official, I would like to see C++14 to be our new
standard.

Thanks,

r-a-v-a-s, 2016-09-09:
We eliminated several memory leaks.
It could have also been caused by a plugin event loop 
preventing LibreCAD from actually terminating.
https://github.com/LibreCAD/LibreCAD/issues/596

Feedback is needed.

strk, 2016-09-09:
I'm not in a condition to be able to test, don't wait for me.
Thanks for your work !

closed 2016-09-09 by r-a-v-a-s