LibreCAD/LibreCAD #689: little feature request for coordinates in command line

billic, 2016-03-22:
a lot of commands have a number of arguments, p.ex. line have at least two XY coordinates.

Is there any option to pass all of this coordinates in one line? A kind of separator inside line or parsing \n in copy&pasted text.

It would be great for user: start a command and then pass a long list of coordinates instead of printing new line for every one point.

This is also would be useful for insert block action: some user program may generate a hundred on points, where user need an object and user simply copying one line like this:
0.256,0;0.256,6;0.512,12;....

Alternatively user can copy many lines with \n as separator and command line takes them all at one time?

in my 2.0.9 any "\n" in pasted text ignored and text
a, b
c, d
inserts as
a, b c, d

I tried to find some request like this but did not find anything, Sorry if there was another one.

r-a-v-a-s, 2016-03-22:
The only similar request was to allow for reading 'command files',
which would just be librecad commands in a file, with some specified delimiter.

r-a-v-a-s, 2016-07-13:
Please see https://github.com/LibreCAD/LibreCAD/issues/782

closed 2016-08-17 by r-a-v-a-s