LibreCAD/LibreCAD #690: New symbols for "unicode.lff"

Symbian9, 2016-03-23:
Here are 4 new symbols for "unicode.lff"
<code>[0406] І
0,0;0,9.

[0407] Ї
1.5,0;1.5,9.
0,11.5;0,12.
3.,11.5;3.,12.

[0456] і
0,0;0,6.
0,8.5;0,9.

[0457] ї
1.5,0;1.5,6.
0,8.5;0,9.
3.,8.5;3.,9.</code>
Add them to https://github.com/LibreCAD/LibreCAD/blob/master/librecad/support/fonts/unicode.lff

dxli, 2016-03-24:
@Symbian9 

Thanks a lot!

Can you send in a pull request?

Symbian9, 2016-03-24:
Cant make a pull, because I have only Symbian phone that can't fully work with github...

closed 2016-03-27 by dxli

Evil-Spirit, 2016-04-07:
There are may be a bug. Missed newlines.

Symbian9, 2021-01-12:
> There are may be a bug. Missed newlines.

@Evil-Spirit, is it still an issue here?

Symbian9, 2021-05-20:
@dxli @lordofbikes, Please, check comments by @Evil-Spirit on some possible issues with `є` and `Ї` characters in `unicode.lff`:

- https://github.com/LibreCAD/LibreCAD/commit/ca6f39feb916390596c7ed3a0cfd88e935cd4e74#commitcomment-17007934
- https://github.com/LibreCAD/LibreCAD/commit/ca6f39feb916390596c7ed3a0cfd88e935cd4e74#commitcomment-17007980