LibreCAD/LibreCAD #691: SegFault when calling a member function of document_interface in Qt slots

Xia-Weiwen, 2016-03-24:
Hi, all.
I have got a problem when I am trying to make a LibreCAD plugin with GUI. I just get a widget and a pushbutton on it. A connection is made like
```C++
connect(ui->pushbutton, SIGNAL(clicked()), this, SLOT(mySlot()));
```
In function mySlot(), I call a member function of document_interface class. The program will crash whenever it goes to this line and whatever this function is. But if the member function is NOT called in a slot, everything is fine.
The code is quite simple, and I write another simple demo plugin to confirm this problem.
The code is like
```C++
class myPlugin : public  QDialog, QC_PluginInterface
{
    Q_OBJECT
    Q_INTERFACES(QC_PluginInterface)
#if QT_VERSION >= 0x050000
    Q_PLUGIN_METADATA(IID "org.librecad.sample" FILE  "sample.json")
#endif

public:
    virtual PluginCapabilities getCapabilities() const;
    virtual QString name() const;
    virtual void execComm(Document_Interface *doc,
                          QWidget *parent, QString cmd);
    explicit myPlugin(QWidget *parent = 0);
    ~myPlugin();
private slots:
    void mySlot();
private:
    Ui::myPlugin *ui;
}
```
```C++
myPlugin::myPlugin(QWidget *parent) :
    QDialog(parent),
    ui(new Ui::myPlugin), QC_PluginInterface()
{
    setWindowFlags(Qt::WindowStaysOnTopHint);
    ui->setupUi(this);
    connect(ui->pushbutton, SIGNAL(clicked()), this, SLOT(mySlot()));
}
```
```C++
void myPlugin::execComm(Document_Interface *doc,QWidget *parent, QString cmd)
{
    Q_UNUSED(doc);
    Q_UNUSED(cmd);
    Q_UNUSED(parent);
    doc1 = doc; // doc1 is defined in myPlugin.h
    show();
    return;
}
```
```C++
void myPlugin::mySlot()
{
        QList<Plug_Entity *> entity_list;
        bool visible=false;
        doc1->getEnt("Please select"); // crash here
        doc1->getAllEntities(&entity_list, visible); // also cause crash
}
```
I debugged and used if(doc1 != 0) to make sure doc1 is not a null pointer.
If I don't use connect() and use mouse right-click -> "go to slot" in Qt Designer instead, I get something like
```C++
void myPlugin::on_pushbutton_clicked()
{
    // the same code as in mySlot()
}
```
and crash also happens.
It seems that I cannot use document_interface function in slots.
Does anyone know the reason? Is this a known bug in librecad or in Qt? Does anyone know how to solve this problem?
Thank you in advance!

r-a-v-a-s, 2016-03-24:
I don't have enough experience to give you a solid answer;
however, you might want to note the difference between your code and `sample.h`.
I see that it has `QC_PluginInterface` in a different class than the `QDialog`.

dxli, 2016-03-25:
@Xia-Weiwen 

Can you share the source code to show the problem?

Xia-Weiwen, 2016-03-28:
@dxli 
Sorry for late reply.
My code is almost the same as what I have posted above. Only names of class and functions are different.
What @r-a-v-a-s has mentioned is very helpful.
The plugin class can only inherit QObject, so the UI or QDialog should be implemented in another class.

Xia-Weiwen, 2016-03-28:
@r-a-v-a-s 
Sorry for late reply.
Thank you! What you said is very helpful.
The plugin class can only inherit QObject, so the UI or QDialog should be implemented in another class.
I changed my code and no crash any more.

closed 2016-03-28 by dxli