LibreCAD/LibreCAD #692: v2.x on Wheezy

JonnyTech, 2016-03-27:
Is it possible to install/build LibreCAD 2.x on Debian Wheezy? Currently running LinuxCNC and can only add v1.0.2.

lordofbikes, 2016-03-27:
I think this is not an easy task as LinuxCNC is based on the outdated Wheezy.

V1.0.2 is the latest version by Wheezy packet manager. There is no package in Wheezy-Backports or -Updates.

I think you have to try to build from source, but this may be tricky as Wheezy's Qt version is 4.6.3. In Wheezy-Backports you should find Qt version 4.8.6. 
I can't test now, which LibreCAD release will build with these Qt versions. There may be issues with c++11 configuration. You can try to checkout older releases, maybe 2.0.4 or 2.0.5 can be build if latest release doesn't work.
Our [Wiki - Build from Source](http://wiki.librecad.org/index.php/Build_from_source) was updated recently to use Qt 5, maybe you find some useful instructions in the history of this site.

r-a-v-a-s, 2016-03-28:
Qt does offer "linux installers".
Although I have no experience using those.

http://www.qt.io/download-open-source/#section-2

r-a-v-a-s, 2016-03-28:
http://doc.qt.io/qt-5/linux.html

JonnyTech, 2016-03-28:
Thanks for the comprehensive reply - trying but getting nowhere yet, I'll report back if I ever get it working.

closed 2016-03-28 by JonnyTech