LibreCAD/LibreCAD #694: Export makes empty jpg tif pdf

schubi71, 2016-04-02:
Sorry, but its annoying and a waste of time to go on with such basically not working program, 
but I want to inform you.
Draw some circle with approx 16 cm Diameter and export it as jpg or tif -> will work.
Put some points in this circuit which you can see and you can also save as .dxf, ok.
And now export again as jpg or tif or pdf -> all empty...

dxli, 2016-04-02:
Just wondering whether you followed lineWidth setting and PrintPreview.

Anyway, it works for me.

Regards,

dxli

schubi71, 2016-04-02:
Dear Dongxu,

its rather simple to say it works for me.
Its not working in the right way, if I can draw at first hand some 
circle AND
export it as "jpg" and get some "jpg" WITH this circle
and then I go on and make some points in this circle and export it again
and there is NOTHING in this jpg.

I am Java Software Engineer and tester and I know when I got some issue
in software or in handling.
Maybe you never installed your software on a clean PC and checked the
first settings and their behaviour after a few steps like desribed.
Or make at least a really clean deinstallation and then try again.
There should be no reason for such a result if I did not change anything
in the programs settings.
I draw this circle without any changing in the new installed program
(which shows its rather intuitive useable), then I first stored it as jpg
and got the shown result as "jpg", ok.
Then I putted some points within the circle and I also exported it, but now
as "tif" with higher resolution (300) (jpg was a little "flickering" 
around the
circle-line).
So why at first hand get some result and then not anymore?
without the use of changing of any of the settings given at the first 
start of
this program.
Yes, its quite easy to put the state of my hint on "closed".

Regards
Kurt Schubert
(First contact with computers and software in 1983
Assembler since about 1990
Unix since 1997
VB6 since 2003
Java since 2007
ABAP since 2010
C++ since 2013)

dxli, 2016-04-02:
Please tell us how to reproduce this bug (not on my machine, but under your environment).

Without the details, there's no much developers can do.

As developers, the "it doesn't work" only bug report is not much helpful. http://forum.librecad.org/How-to-report-bugs-and-request-support-td5713393.html

Please provide all details:

0, start LibreCAD (version, OS);
1, Main menu: Draw -> Circle -> Center Point;
2, when prompted "Specify Circle Center", type in 0, 0 as center
3, ...

r-a-v-a-s, 2016-04-02:
Relevant information would be your operating system and what version of the program you are using.

dxli, 2016-04-02:
@r-a-v-a-s 

our wiki doesn't have enough on printing/exporting.

print preview is tricky. We talked about warning of unreasonable scales: 1, page too small; 2, page too big; 3, default image exporting parameters are not "smart".

r-a-v-a-s, 2016-04-02:
I can't reproduce the problem.

The paper format option needs to be set in Current Drawing Preferences for proper PDF export.

Users need to be prompted to set that option, 
and to deal with the print preview window.

schubi71, 2016-04-03:
Hello Dongxu,

I just read your answer (here in Germany its now Sunday about 10am).
I will answer you later to find some helpful way to fix the problem.

Best regards
Kurt

schubi71, 2016-04-03:
Hello Dongxu,

I just made some complete deinstallation and installed it again, now it works,
I get a filled "jpg" and a filled "tif" including circle AND points when I export the image
the second time so the rest is now to play with the program and to try the settings.
Dont know what happened and where this mistake came from and hope not to find it again.
(W7 64bit LibreVersion  2.09)
Sorry for my angry words and for the spending of your time.
Best greets for Sunday!
schubi71
(For the moment I see it as solved, so you can close this.)

schubi71, 2016-04-03:
Dear Dongxu and ravas,

facing the same problem again.
Putted the relating files into some zip in attach.
W7-64k LibreCAD 2.09 fresh installation/engl/engl

Make some circle - exporting as jpg and gif -> ok
Adding some points within the circle - exporting as jpg and gif -> 
receiving empty files

As already told, I reinstalled it and repeated the whole process (11:00am)
it looked like ok and I stated this on the forum

Now due to some bad setted point I drawed again a fresh circle with 
points in it -> problems again
(empty jpg and tif)

Sorry, maybe you understand this...

Best regards
Kurt

r-a-v-a-s, 2016-04-03:
If setting the paper format (e.g. A4) doesn't help,
then all I can do is offer a work around.

Since the LibreCAD export produces rather unattractive jpg files,
I export as MakerCAM SVG and then open it in InkScape (also free / open-source)
and then export. You might need to tweak the stroke width and scale the objects,
but it produces properly anti-aliased circles.

[image]

schubi71, 2016-04-03:
> g4221

Hello ravas,

I looked for this paper format, which is in Edit -> Current Drawing 
Preferences
Its already set for "A4" which seems to be from the start because I did not
go to this menue point before...
Which really is mysterious is the fact that at first time when I draw 
just the circuit
I get the jpg export in a normal way and then later with those mentioned 
points in the circuit
I cannot export it anymore, better say the jpg and the tif is empty.

If you load the "Circle_03.dxf" from my zip-file into your LibreCAD on your
PC, can you export this as jpg and as tif (resolution 300)?

schubi71, 2016-04-03:
Hello ravas,

I got it!
When I export the image the first time the width and height stays on for 
example 320,
even if I adapted the resolution.

If I repeat this procedure the system suddenly starts to calculate a new 
size of
the picture which is now suddenly 96121 (and much to big).
It could get more obvious if you write just something like (px) behind the
numbers of  height and width.

But this is an error in program.

Either you do this size calculation from the start or never...

Regards
Kurt

r-a-v-a-s, 2016-05-30:
I just found that there was a bad signal connection.
Now when you set the resolution it will automatically change the width / height.
Try lower values first... such as 1,2 ,3...