LibreCAD/LibreCAD #695: Windows installer creates desktop shortcut in current users desktop folder

websafe, 2016-04-04:
The Windows installer creates the desktop shortcut in the current users desktop folder instead of `%PUBLIC%\Desktop`.

lordofbikes, 2016-04-04:
I have to check, if there are any side effects with user settings.
Then I'd prefer a solution to let the user decide where to create the short cut, installing for the current or all users.

websafe, 2016-04-08:
Yes, let the user decide but if it's possible please allow silent installers to choose `%PUBLIC%\Desktop` for desktop icons, because otherwise installing on multiple machines via `chocolatey` requires additional scripting.

In corporate environments users have no privileges to install software so admins have to move this shortcut every time to `%PUBLIC%\Desktop` and after each upgrade an additional shortcut on administrators desktop is created, which needs to be removed :-)

You should probably also check the discussion on https://chocolatey.org/packages/librecad :-)

r-a-v-a-s, 2016-04-08:
http://forum.librecad.org/Windows-installer-sends-users-to-the-donation-page-td5712319.html