LibreCAD/LibreCAD #696: Compile error: CDataSen::Serialize should be defined with "const"

pgovind, 2016-04-11:
"jwwdoc.cpp" defines the function CDataSen::Serialize. On _MSC_VER, the declaration makes the function const. However, the definition is not const and this leads to compile errors. I made a local change to make it const, but there are other error messages on visual studio, the first of which is on the next line :

    ofstr	<< (double)m_start.x << (double)m_start.y
        << (double)m_end.x << (double)m_end.y;

where the compiler complains that more than one operator "<<" is defined for operands std::ofstream << double. 

Not sure how others are not seeing this. Is nobody else on windows or am I doing something wrong?

r-a-v-a-s, 2016-04-12:
We use the MinGW version of Qt.

r-a-v-a-s, 2016-05-03:
I don't think anyone is opposed to supporting msvc;
dxli and I don't use Windows,
and lordofbikes probably isn't interested enough.

Feel free to submit a patch.

rvt, 2016-11-27:
We can remove jww support if we want to.

shawncurry, 2019-05-11:
I have fixed all of the MSVC toolchain issues, including this one.  https://github.com/LibreCAD/LibreCAD/pull/1096

I managed to preserve the jww support in the process.  I can still import all of the jww files I found included in the jwcad project, with the exception of the ones containing Japanese characters in the filenames.  The current release also has a problem with these ones, likely a separate filename handling issue.

sand1024, 2026-09-26:
non relevant, fixed long time ago

closed 2026-09-26 by sand1024